huggingface/lerobot #982: MultiDataset

banban3forever, 2025-04-15:


banban3forever, 2025-04-15:
I am using lerobot pi0 to train a multi-task policy. Will it work if i provide a list of repo_id and remove the exception part of make_dataset.

banban3forever, 2025-07-23:
> [@banban3forever](https://github.com/banban3forever) did you solve this problem?

No, i didn't try this way because you can easily mix data of different tasks when convert data to lerobot format. You can try the scripts mentioned [here](https://github.com/Physical-Intelligence/openpi/blob/main/examples/aloha_real/convert_aloha_data_to_lerobot.py).

pkooij, 2026-10-21:
Thanks! We’re keeping multi-dataset training requests in #1576. Compatible datasets can be merged as a workaround; direct multi-dataset training remains separate work. Closing this duplicate.

closed 2026-10-21 by pkooij