huggingface/lerobot #985: Calibration tutorial

amisev, 2025-04-15:
I've noticed a potential inconsistency in the robot's wrist orientation between the documentation images. In the assembly guide image (assembly)[https://github.com/huggingface/lerobot/blob/main/examples/10_use_so100.md#d-step-by-step-assembly-instructions] (

[image]

), the wrist appears to face towards the motor. However, in the calibration image (

[image]

) (calibration)[https://github.com/huggingface/lerobot/blob/main/examples/10_use_so100.md#e-calibrate], the wrist seems to be oriented in the opposite direction (facing away from the motor). Is this difference intentional for the calibration step, or could this be an error in one of the images?

Usimian, 2025-04-27:
So which way should it go?  Does it matter?

Lottie56, 2025-05-05:
I've seen same inconsistency, I've finally decided to use 2nd image as reference.