huggingface/lerobot #1000: How to implement a new policy?

Elycyx, 2025-04-19:
How can I integrate a new policy (e.g., OpenVLA) into LeRobot, and specifically, which files do I need to modify?

helper2424, 2025-04-19:
@Elycyx wow, thats sounds great.

Check the following folder - https://github.com/huggingface/lerobot/tree/main/lerobot/common/policies.

You need to add `modeling` and `configuration` parts. The model should inherit from `PreTrainedPolicy`.

helper2424, 2025-04-19:
if u need any help - feel free ping, but in general it should be pretty straightforward.

Elycyx, 2025-04-19:
> if u need any help - feel free ping, but in general it should be pretty straightforward.

Thank you so much, I'll give it a try.😀

jjiwon-kim, 2025-07-21:
Hi, @Elycyx ! 

How was implementing Openvla as lerobot policy? Have you finished it?   
Just wondering since I'm about to start doing the same thing. 😄