huggingface/lerobot #1010: [TxRxResult] Incorrect status packet!

jlhausherr, 2025-04-21:
Hello,
I have just encountered the problem 
_Torque_Enable_shoulder_pan_shoulder_lift_elbow_flex_wrist_flex_wrist_roll_gripper: [TxRxResult] Incorrect status packet!_
In my case it wasn't a connection problem.
The supplier in China who refers to his product as the SO100 has supplied me with the wrong STS3215 motors.
My motors are in 7.4v which puts them in overvoltage alarm when I power them and 12v (small red LED flashing on each motor). 
My motors do not have the correct firmware version, they are in version 3.9 when version 3.10 is required
One of these cases is enough for the calibration script to return the error [TxRxResult] Incorrect status packet!
Feetech provides a motor control utility. (FT SCServo Debug)
It can be used to check that the connections are OK and to update the firmware.
After updating the firmware and powering the motors at 7.4V I no longer get the error.
Hope this helps.

[image]

SangbumChoi, 2025-05-16:
@jlhausherr Facing same error but im mac user how can I fix this?

SangbumChoi, 2025-05-17:
https://github.com/Kotakku/FT_SCServo_Debug_Qt

SangbumChoi, 2025-05-17:
https://github.com/Adam-Software/Feetech-Servo-SDK/blob/5d25bbe8300858a032ee568e5abed2a184f1d840/scservo_sdk/scservo_def.py#L24

rhysdg, 2025-05-17:
Hey there all! Just a confirmation on my end that the fix suggested by  @jlhausherr worked with  Ubuntu 22.04:

- Motor id assignment on Ubuntu with Lerobot's so101 tutorial
- Checked for successful communication with the tool suggested by @SangbumChoi 
- Booted to a Windows partition to use Feetech's official debug tool and upgraded the firmware for all motors
- Boot back to Ubuntu and calibration now works for bot the leader and the follower!

Cheers for all the tips and excited to see what I can pull off!

SangbumChoi, 2025-05-18:
@rhysdg  any suggestion for mac user?

SangbumChoi, 2025-05-19:
[image]

well the code identify the port (e.g. Com) but cannot select it :(. BTW thanks for the kind description

SangbumChoi, 2025-05-19:
@CarolinePascal Suggested branch fix the issue of setting the Com (which i think it is the port) but it automatically returns back and could not detect the motor :(

BTW my PC is mac m3

SangbumChoi, 2025-05-19:
Actually only one motor was in the firmware version 3.9 :)

But don't know how to update this...

[image]

CarolinePascal, 2025-05-19:
So far, the only _easy_ way to upgrade the firmware of Feetech servos is to use (Feetech's utility)[https://gitee.com/ftservo/fddebug/tree/master] on a Windows virtual machine. On MacOS, I've successfully tested Parallels (https://www.parallels.com) and I think similar results may be achieved on Linux with [Virtual Box](https://www.virtualbox.org).

On a side note, @aliberts confirmed that a single servo with a different firmware could be the cause of your problem !

Let us know if you find a better way to upgrade Feetech's firmware on MacOS.

Best,

Caroline.

SangbumChoi, 2025-05-20:
Thanks again for the kind description I will try with Window PC and let you know if the problem has been fixed!

aliberts, 2025-05-20:
> On a side note, @aliberts confirmed that a single servo with a different firmware could be the cause of your problem !

Regarding this, we already have [a check](https://github.com/huggingface/lerobot/blob/05dfa26c541542818932536c860214de0775dee0/lerobot/common/motors/feetech/feetech.py#L155-L161) in #777 to make sure all servos connected to the bus have the same firmware version and raise an error if not. This will be merged soon.

closed 2025-05-20 by CarolinePascal

SangbumChoi, 2025-05-21:
FYI I think version of 3.9 is not the problem. The real problem is that all 6 motors should have the same version such as 3.9 or 3.10.

jonasschramm, 2025-05-27:
@jlhausherr Thanks for sharing! I hope this will fix my error as well. Which servo driver board did you use for the firmware update?

SangbumChoi, 2025-05-28:
3.9, 3.10 will both also work just make it all same

Kracozebr, 2025-06-14:
In my case leader firmware version for all servo was 3.10 and on follower was 3.9 and that leads to problems in teleporting. Some servo rotate in the opposite direction. Changing firmware to 3.10 fixed that

cverrier, 2025-06-21:
[For future reference]

I am on macOS 15.5 (ARM architecture, MacBook Pro M1), and I followed [the official SO-101 tutorial](https://huggingface.co/docs/lerobot/so101) to calibrate my SO-101 follower arm. I created a Python 3.10 virtual environment (`uv venv --python 3.10 && uv sync --all-extras`) and I ran the following Python script (commit c940676bdda5ab92e3f9446a72fafca5c550b505):
```zsh
python -m lerobot.calibrate \
    --robot.type=so101_follower \
    --robot.port=/dev/tty.usbmodem5A680097201 \
    --robot.id=follower
```

The first time, I got the following error:
```
INFO 2025-06-21 14:37:27 calibrate.py:71 {'robot': {'calibration_dir': None,
           'cameras': {},
           'disable_torque_on_disconnect': True,
           'id': 'follower',
           'max_relative_target': None,
           'port': '/dev/tty.usbmodem5A680097201',
           'use_degrees': False},
 'teleop': None}
INFO 2025-06-21 14:37:27 follower.py:101 follower SO101Follower connected.
INFO 2025-06-21 14:37:27 follower.py:108
Running calibration of follower SO101Follower
Traceback (most recent call last):
  File "/Users/clementverrier/.local/share/uv/python/cpython-3.10.18-macos-aarch64-none/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/Users/clementverrier/.local/share/uv/python/cpython-3.10.18-macos-aarch64-none/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/Users/clementverrier/repos/lerobot/lerobot/calibrate.py", line 84, in <module>
    calibrate()
  File "/Users/clementverrier/repos/lerobot/.venv/lib/python3.10/site-packages/draccus/argparsing.py", line 225, in wrapper_inner
    response = fn(cfg, *args, **kwargs)
  File "/Users/clementverrier/repos/lerobot/lerobot/calibrate.py", line 79, in calibrate
    device.calibrate()
  File "/Users/clementverrier/repos/lerobot/lerobot/common/robots/so101_follower/so101_follower.py", line 109, in calibrate
    self.bus.disable_torque()
  File "/Users/clementverrier/repos/lerobot/lerobot/common/motors/feetech/feetech.py", line 297, in disable_torque
    self.write("Torque_Enable", motor, TorqueMode.DISABLED.value, num_retry=num_retry)
  File "/Users/clementverrier/repos/lerobot/lerobot/common/motors/motors_bus.py", line 1022, in write
    self._write(addr, length, id_, value, num_retry=num_retry, raise_on_error=True, err_msg=err_msg)
  File "/Users/clementverrier/repos/lerobot/lerobot/common/motors/motors_bus.py", line 1046, in _write
    raise ConnectionError(f"{err_msg} {self.packet_handler.getTxRxResult(comm)}")
ConnectionError: Failed to write 'Torque_Enable' on id_=1 with '0' after 1 tries. [TxRxResult] Incorrect status packet!
```

Then I unplugged the SO-101 follower arm from my computer, replugged it and ran the exact same script again, and everything worked. I haven't tried to run the script another time without unplugging the follower arm.

DominikFilipiak, 2026-08-31:
I had this problem as well - what helped me was running these commands **before** (re-) calibrating leader and follower.
```sh
sudo chmod 666 /dev/your-follower
sudo chmod 666 /dev/your-leader
```
I got the bug because I first calibrated both arms without chmod. I did the chmod afterwards, but that resulted in `Failed to write 'Torque_Enable' on id_=1 with '0' after 6 tries. [TxRxResult] There is no status packet!` during the teleoperation.