huggingface/lerobot #1026: Manual calibration is buggy. Please enable auto-calibration for so-100 arm

odellus, 2025-04-24:
### System Info

```Shell
I'm using lerobot from main branch about a week ago (don't see any releases on github)
Ubuntu 22.04 OS.
Here are my arm configurations
https://gist.github.com/odellus/b26f2a781ca74a29b4a33c77a7b5cf70
https://gist.github.com/odellus/82a72e6299716ddcbb6eaedfcc5bef16
```

### Information

- [x] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

Steps to reproduce behavior:

1. put together the robot
2. try to calibrate the robot
3. get it working
4. have it break for no obvious reason
5. receive error messages from not being able to calibrate the robot again

I'm getting behavior similar to https://github.com/huggingface/lerobot/issues/672
and also #930 . Tried to take advice of "don't worry about it, it settles down after a second" but when I tried to record a dataset everything broke and now my arm is completely non-functional. 

### Expected behavior

I'd hope that once I have the robot calibrated that the behavior would be sort of reproducible and consistent. Not working one hour and then irreparably broken the next.

Ideally I wouldn't have to fight with manual calibration at all and the arm would have an auto-calibration step as is standard.

odellus, 2025-04-26:
I was able to fix the issue by taking apart the 3rd servo joint on both arms and running the servo ID command

```bash
uv run lerobot/scripts/configure_motor.py \
  --port /dev/ttyACM0
  --brand feetech \
  --model sts3215 \
  --baudrate 1000000 \
  --ID 3
```
to recenter the drive and reinstall the rest of the manipulator with the horn in the correct position. This seems tedious. Would be great to get automatic calibration working for so100 arm.

dhpitt, 2025-05-02:
Bumping this, the 3rd servo joint is a repeated pain point for me as well.

closed 2025-11-13 by github-actions[bot]