huggingface/lerobot #1069: In lerobot/visualize_dataset, the curve of leader_arm cannot be seen and only the follower_arm (observation) is visible.

ZHHhang, 2025-05-05:
### System Info

```Shell
Previously, I was able to view all data from my uploaded dataset in visualize_dataset, including the joint angles of both the leader_arm and the follower_arm (observation). However, I recently noticed that the joint angles of the leader_arm are no longer visible. This information is very important to my work.

I would like to ask whether this change is due to a recent update to the website, or if there is a specific action I need to take in order to include the leader_arm joint angles in the visualization.

I believe this curve information is also essential for many researchers. I would be very grateful if the maintainers could look into this issue and restore the visibility of the leader_arm data.
```

### Information

- [x] One of the scripts in the examples/ folder of LeRobot
- [x] My own task or dataset (give details below)

### Reproduction

Steps to reproduce the issue:

Upload a dataset that includes both leader_arm and follower_arm data.

Open the dataset in https://huggingface.co/spaces/lerobot/visualize_datase)

Observe that only follower_arm joint angles are shown in the visualization.

### Expected behavior

The joint angle curves for both leader_arm and follower_arm should be visible, as they were previously.
If this change is due to a recent update, could you please clarify whether it’s intentional or a bug?
If it requires a specific action on my end to display leader_arm data again, I would appreciate your guidance.
I would be very grateful if this issue could be addressed or clarified.

mishig25, 2025-05-05:
looking at it

mishig25, 2025-05-05:
> Upload a dataset that includes both leader_arm and follower_arm data.

do you have an example dataset on hf hub that I can test on?

mishig25, 2025-05-05:
@ZHHhang thanks a lot for notifying about it. the issue is fixed now ✅