huggingface/lerobot #1072: How to merge collected data into one?

milong26, 2025-05-06:
For stability I collect data 10 episode by 10. Then forming this:
repo_id/first,repo_id_second...
I want to merge them together to repo_id/one_task for training, but it's hard to fix meta files. 

I'm not sure if this approach helps with training, or if I should determine the number of episodes needed for training in advance when collecting data.

milong26, 2025-05-06:
e.g. in repo_id/first,it has episode_000000~episode_000009.in repo_id/second, it has episode_000000~episode_000007...

i want to get a new folder repo_id/new with episode_000000~episode_000009~episode_000016~...

Tavish9, 2025-05-06:
Hi, @milong26 please refer to [any4lerobot](https://github.com/Tavish9/any4lerobot) for details. 

We have implemented a merging uitls

milong26, 2025-05-07:
> Hi, [@milong26](https://github.com/milong26) please refer to [any4lerobot](https://github.com/Tavish9/any4lerobot) for details.
> 
> We have implemented a merging uitls

Thank you so much!!

closed 2025-05-07 by milong26