huggingface/lerobot #1078: Running a model trained earlier leads to stats error

Himhawkins, 2025-05-06:
### System Info

```Shell
Error: File "/home/pranav/Repo/lerobot/lerobot/common/policies/normalize.py", line 169, in forward
    assert not torch.isinf(mean).any(), _no_stats_error_str("mean")
AssertionError: `mean` is infinity. You should either initialize with `stats` as an argument, or use a pretrained model.
 Can someone help with resolving this? I am getting this while running evaluate on an ACT model i trained. Model: ps5387/SHAKE29k
```

### Information

- [ ] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

Download the ps5387/SHAKE29k and running python lerobot/scripts/control_robot.py \
  --robot.type=so100 \
  --control.type=record \
  --control.fps=30 \
  --control.single_task="Grab and Shake hand when you see it" \
  --control.repo_id=ps5387/eval_act_so100_shake2_1 \
  --control.tags='["so100","shake"]' \
  --control.warmup_time_s=5 \
  --control.episode_time_s=25 \
  --control.reset_time_s=10 \
  --control.num_episodes=10 \
  --control.push_to_hub=true \
  --control.policy.path=shake20k/pretrained_model

### Expected behavior

Getting error File "/home/pranav/Repo/lerobot/lerobot/common/policies/normalize.py", line 169, in forward
    assert not torch.isinf(mean).any(), _no_stats_error_str("mean")
AssertionError: `mean` is infinity. You should either initialize with `stats` as an argument, or use a pretrained model.

CarolinePascal, 2025-05-14:
HI @Himhawkins !

Could you check if the naming issue reported in #1095 solves your issue as well ?

Best,

Caroline.

yangfengzzz, 2025-06-29:
> HI [@Himhawkins](https://github.com/Himhawkins) !
> 
> Could you check if the naming issue reported in [#1095](https://github.com/huggingface/lerobot/issues/1095) solves your issue as well ?
> 
> Best,
> 
> Caroline.

Should I re-record dataset in new lerobot? I found there is a large code refactor in the July.