huggingface/lerobot #1083: select() timeout. Error reading in thread: Can't capture color image from camera 6.

milong26, 2025-05-08:
In SOarm100 experiment, I met this problem sometimes:(camera 6 is wrist camera)

```
[ WARN:0@26.501] global cap_v4l.cpp:1048 tryIoctl VIDEOIO(V4L2:/dev/video6): select() timeout.
Error reading in thread: Can't capture color image from camera 6.
```

After I saw this WARN I knew these:

1. I quit this run and `python lerobot/common/robot_devices/cameras/opencv.py     --images-dir outputs/images_from_opencv_cameras` directly, it can catch color image from camera 6 and save them, so not cable lost.
2. When I check the images saved from wrist 6(the problem never happened very early),episode 0 ok,if this WARN came out when recording episode 1, the images in folder `episode_000001` may show like this: images 0-n goes well but from n+1 to end, all the images are same :(
3. if i don't quit at episode 1, episode 2,3,4... will show the same result as the same image in episode 1(the image from n+1 to end).
4. the index number of all episodes are right, for `scene` folder has so many images too.

Maybe it's the problem of bandwidth because when I don't use scene camera, the WARN never shows and the outputs are fine. But I only record 640*480*3, wrist camera using usb2.0 while scene camera use 3.0 linking to different bus. I don't think bandwidth will make too much difference to this :(

If it's the problem of bandwidth,why images from n+1 to end and next episodes shows same? I think it's about image saved buffer or thread? Can anyone help with this?  To me if I can't avoid seeing this WARN,at least I can only see same images from n to n+1, not n+1 to end. 

Thanks! tell me if you need any more info!

milong26, 2025-05-09:
I use `read` instead of `async_read` and use longer reset_time now no WARN xD

Btw I may find why when `WARN` showed up the images were all same, when I used `read` with short reset_time, it shows `OSerror can't capture color image` and exit. But when in thread, the program would not exit so the images can only be same.

But i don't know how to fix this :[ I thought when use `async_read`+long reset time could fix this but I failed. So I use just `read`. If anyone know how to use `async/_read` without seeing `WARN` please tell me.

CarolinePascal, 2025-05-09:
Hi again @milong26 !

This does sound like a bandwidth issue to me, as issues involving USB 2.0 have already been reported on our [Discord](https://t.co/Sx2jdT0jeF). I would advise to check if your camera is able to reliably provide frames at the requested FPS specified in [`configs.py`](https://github.com/huggingface/lerobot/blob/main/lerobot/common/robot_devices/robots/configs.py). Here is a small python script to do so (thks @imstevenpmwork) :

```python
import argparse
import statistics
import time
import sys
import cv2

def parse_arguments():
    """Parse command-line arguments."""
    parser = argparse.ArgumentParser(
        description="Capture video from a webcam and display FPS statistics."
    )
    parser.add_argument("--camera", type=int, default=0,
                        help="Index of the target camera (default: 0)")
    parser.add_argument("--width", type=int, default=640,
                        help="Target video width (default: 640)")
    parser.add_argument("--height", type=int, default=480,
                        help="Target video height (default: 480)")
    parser.add_argument("--fps", type=int, default=60,
                        help="Target frames per second (default: 60)")
    parser.add_argument("--collect_time", type=int, default=2,
                        help="Interval in seconds to collect FPS data (default: 1)")
    parser.add_argument("--duration", type=int, default=60,
                        help="Total duration in seconds for the test (default: 10)")
    return parser.parse_args()

def initialize_camera(args):
    """Initialize the camera with the provided arguments and return the VideoCapture object."""
    vc = cv2.VideoCapture(args.camera, cv2.CAP_ANY)
    if not vc.isOpened():
        sys.exit("Error: Unable to open the camera.")
    
    # Set camera properties
    vc.set(cv2.CAP_PROP_FRAME_WIDTH, args.width)
    vc.set(cv2.CAP_PROP_FRAME_HEIGHT, args.height)
    vc.set(cv2.CAP_PROP_FPS, args.fps)
    vc.set(cv2.CAP_PROP_BUFFERSIZE, 1)
    
    return vc

def print_camera_info(vc, args):
    """Display requested and actual camera settings."""
    actual_width = vc.get(cv2.CAP_PROP_FRAME_WIDTH)
    actual_height = vc.get(cv2.CAP_PROP_FRAME_HEIGHT)
    actual_fps = vc.get(cv2.CAP_PROP_FPS)
    
    print(f"Requested: {args.width}x{args.height} @ {args.fps}fps")
    print(f"Actual: {int(actual_width)}x{int(actual_height)} @ {actual_fps:.1f}fps")

def capture_and_measure(vc, collect_time, duration):
    """
    Capture frames from the camera, measure FPS over each collection interval, 
    and return the list of measured FPS values.
    """
    fps_values = []
    frame_count = 0
    prev_time = time.time()
    start_time = prev_time

    try:
        while time.time() - start_time < duration:
            ret, frame = vc.read()
            if not ret:
                print("Warning: Failed to capture frame.")
                break

            # Display the captured frame (optional)
            print(frame.shape)
            cv2.imshow("preview", frame)
            cv2.waitKey(1)

            frame_count += 1
            current_time = time.time()
            elapsed_time = current_time - prev_time

            # Once the collection period has passed, compute the FPS for that interval
            if elapsed_time >= collect_time:
                fps = frame_count / elapsed_time
                print(f"FPS: {fps:.2f}")
                fps_values.append(fps)
                frame_count = 0
                prev_time = current_time

    except KeyboardInterrupt:
        print("Capture interrupted by user.")

    return fps_values

def display_statistics(fps_values):
    """Display statistics for the collected FPS values."""
    if len(fps_values) > 1:
        average_fps = statistics.mean(fps_values)
        std_dev_fps = statistics.stdev(fps_values)
        print("\n--- FPS Statistics ---")
        print(f"Average FPS: {average_fps:.2f}")
        print(f"Standard Deviation of FPS: {std_dev_fps:.2f}")
    elif fps_values:
        average_fps = statistics.mean(fps_values)
        print("\n--- FPS Statistics ---")
        print(f"Average FPS: {average_fps:.2f}")
        print("Standard Deviation requires at least two data points.")
    else:
        print("\nNo FPS data collected.")

def main():
    args = parse_arguments()
    print(f"Target Camera: {args.camera}")
    print(f"Target Width: {args.width}")
    print(f"Target Height: {args.height}")
    print(f"Target FPS: {args.fps}")
    print(f"Collect Time: {args.collect_time}")
    print(f"Test Duration: {args.duration}")

    vc = initialize_camera(args)
    print_camera_info(vc, args)

    fps_values = capture_and_measure(vc, args.collect_time, args.duration)
    
    vc.release()
    display_statistics(fps_values)

if __name__ == "__main__":
    main()
```

Best,

Caroline.

milong26, 2025-05-11:
> Hi again [@milong26](https://github.com/milong26) !
> 
> This does sound like a bandwidth issue to me, as issues involving USB 2.0 have already been reported on our [Discord](https://t.co/Sx2jdT0jeF). I would advise to check if your camera is able to reliably provide frames at the requested FPS specified in [`configs.py`](https://github.com/huggingface/lerobot/blob/main/lerobot/common/robot_devices/robots/configs.py). Here is a small python script to do so (thks [@imstevenpmwork](https://github.com/imstevenpmwork)) :
> 
> import argparse
> import statistics
> import time
> import sys
> import cv2
> 
> 
> def parse_arguments():
>     """Parse command-line arguments."""
>     parser = argparse.ArgumentParser(
>         description="Capture video from a webcam and display FPS statistics."
>     )
>     parser.add_argument("--camera", type=int, default=0,
>                         help="Index of the target camera (default: 0)")
>     parser.add_argument("--width", type=int, default=640,
>                         help="Target video width (default: 640)")
>     parser.add_argument("--height", type=int, default=480,
>                         help="Target video height (default: 480)")
>     parser.add_argument("--fps", type=int, default=60,
>                         help="Target frames per second (default: 60)")
>     parser.add_argument("--collect_time", type=int, default=2,
>                         help="Interval in seconds to collect FPS data (default: 1)")
>     parser.add_argument("--duration", type=int, default=60,
>                         help="Total duration in seconds for the test (default: 10)")
>     return parser.parse_args()
> 
> 
> def initialize_camera(args):
>     """Initialize the camera with the provided arguments and return the VideoCapture object."""
>     vc = cv2.VideoCapture(args.camera, cv2.CAP_ANY)
>     if not vc.isOpened():
>         sys.exit("Error: Unable to open the camera.")
>     
>     # Set camera properties
>     vc.set(cv2.CAP_PROP_FRAME_WIDTH, args.width)
>     vc.set(cv2.CAP_PROP_FRAME_HEIGHT, args.height)
>     vc.set(cv2.CAP_PROP_FPS, args.fps)
>     vc.set(cv2.CAP_PROP_BUFFERSIZE, 1)
>     
>     return vc
> 
> 
> def print_camera_info(vc, args):
>     """Display requested and actual camera settings."""
>     actual_width = vc.get(cv2.CAP_PROP_FRAME_WIDTH)
>     actual_height = vc.get(cv2.CAP_PROP_FRAME_HEIGHT)
>     actual_fps = vc.get(cv2.CAP_PROP_FPS)
>     
>     print(f"Requested: {args.width}x{args.height} @ {args.fps}fps")
>     print(f"Actual: {int(actual_width)}x{int(actual_height)} @ {actual_fps:.1f}fps")
> 
> 
> def capture_and_measure(vc, collect_time, duration):
>     """
>     Capture frames from the camera, measure FPS over each collection interval, 
>     and return the list of measured FPS values.
>     """
>     fps_values = []
>     frame_count = 0
>     prev_time = time.time()
>     start_time = prev_time
> 
>     try:
>         while time.time() - start_time < duration:
>             ret, frame = vc.read()
>             if not ret:
>                 print("Warning: Failed to capture frame.")
>                 break
> 
>             # Display the captured frame (optional)
>             print(frame.shape)
>             cv2.imshow("preview", frame)
>             cv2.waitKey(1)
> 
>             frame_count += 1
>             current_time = time.time()
>             elapsed_time = current_time - prev_time
> 
>             # Once the collection period has passed, compute the FPS for that interval
>             if elapsed_time >= collect_time:
>                 fps = frame_count / elapsed_time
>                 print(f"FPS: {fps:.2f}")
>                 fps_values.append(fps)
>                 frame_count = 0
>                 prev_time = current_time
> 
>     except KeyboardInterrupt:
>         print("Capture interrupted by user.")
> 
>     return fps_values
> 
> 
> def display_statistics(fps_values):
>     """Display statistics for the collected FPS values."""
>     if len(fps_values) > 1:
>         average_fps = statistics.mean(fps_values)
>         std_dev_fps = statistics.stdev(fps_values)
>         print("\n--- FPS Statistics ---")
>         print(f"Average FPS: {average_fps:.2f}")
>         print(f"Standard Deviation of FPS: {std_dev_fps:.2f}")
>     elif fps_values:
>         average_fps = statistics.mean(fps_values)
>         print("\n--- FPS Statistics ---")
>         print(f"Average FPS: {average_fps:.2f}")
>         print("Standard Deviation requires at least two data points.")
>     else:
>         print("\nNo FPS data collected.")
> 
> 
> def main():
>     args = parse_arguments()
>     print(f"Target Camera: {args.camera}")
>     print(f"Target Width: {args.width}")
>     print(f"Target Height: {args.height}")
>     print(f"Target FPS: {args.fps}")
>     print(f"Collect Time: {args.collect_time}")
>     print(f"Test Duration: {args.duration}")
> 
>     vc = initialize_camera(args)
>     print_camera_info(vc, args)
> 
>     fps_values = capture_and_measure(vc, args.collect_time, args.duration)
>     
>     vc.release()
>     display_statistics(fps_values)
> 
> 
> if __name__ == "__main__":
>     main()
> 
> Best,
> 
> Caroline.

Thanks! I tried this script,it showed:
--- FPS Statistics ---
Average FPS: 28.88
Standard Deviation of FPS: 1.44

I set fps=30 in configs.py. So it seems no problem when only this camera. But to me I use two cameras they caused bandwidth problem (i really don't know why,they even in two different usb bus in `lsusb` result) so I have to use `MJPEG` encoding. Thanks I'll checkit.

milong26, 2025-05-12:
Got it, this problem is caused by my usb cable :( 

Is there any solution to deal with camera lost connection due to bandwidth?
In [opencv,py](https://github.com/huggingface/lerobot/blob/main/lerobot/common/robot_devices/cameras/opencv.py)
```python
if not ret:
            raise OSError(f"Can't capture color image from camera {self.camera_index}.")
```

I tried to re-start camera here but it may cause problem during async_read.

lanseyege, 2025-05-12:
> Got it, this problem is caused by my usb cable :(
> 
> Is there any solution to deal with camera lost connection due to bandwidth? In [opencv,py](https://github.com/huggingface/lerobot/blob/main/lerobot/common/robot_devices/cameras/opencv.py)
> 
> if not ret:
>             raise OSError(f"Can't capture color image from camera {self.camera_index}.")
> I tried to re-start camera here but it may cause problem during async_read.

I meet the same problem ...

CarolinePascal, 2025-05-12:
Hi @milong26 !

As it is currently implemented, `async_read()` will always output a frame, regardless of the output if `read()` (that is what you always get the same frame once the camera disconnects).

If you want to try and reconnect the camera on failure, you identified the correct location in the code (as it should not impact the asynchronous thread execution). If you are still unsure about how to proceed, feel free to seek help on our Discord ! And of course, if you come up with a working solution, please fill a PR so we can include your fix in our codebase ;)

Best,

Caroline.

CarolinePascal, 2025-05-14:
Closing the issue as it was identified as a hardware/bandwidth issue.

closed 2025-05-14 by CarolinePascal

127772828, 2025-09-25:
I don't think it's a hardware/bandwidth issus. Because yesterday, I used a 1080p camera and a claw camera recommended by the official, along with the data cables for the master-slave mechanical arms, a total of four cables were connected to one hub, and the hub was connected to a USB 3.0 port. I adjusted the resolution of the two cameras to 1080p 30Hz, and was able to record the dataset perfectly. However, the problem was that the conda environment had issues later, which led to a reconfiguration of the environment. But in the new environment, this problem occurred, and the cameras could never be connected. But when using the opencv.py file to check the camera index, all image frames could be output normally.

CarolinePascal, 2025-09-29:
Hi @127772828 

Could you please provide more details on your condas environments so that we can help identify the issue ? Also, could you try the debug script provided above in the discussion ?

Best,

Caroline.