huggingface/lerobot #1088: Latest version doesn't support MJPEG

milong26, 2025-05-09:
I try to use `MJPEG` in opencv camera setting:
```python
self.camera.set(cv2.CAP_PROP_FOURCC, cv2.VideoWriter_fourcc(*'MJPG'))
```

This works well in [42bf1e8b](https://github.com/huggingface/lerobot/commit/42bf1e8b) for in [78fd1a1e04f87f3ec52428f72c29485124472fbc](https://github.com/huggingface/lerobot/tree/78fd1a1e04f87f3ec52428f72c29485124472fbc)  support MJPEG image to video but in [6d723c4](https://github.com/huggingface/lerobot/commit/6d723c45a993d1e030dfda20d48a893a74a4fe11) it adds restriction on image format.

Does lerobot require YUYV format?

CarolinePascal, 2025-05-09:
Hi @milong26 !

If I understand your issue correctly, you are trying to use the [`MJPEG`](https://en.wikipedia.org/wiki/Motion_JPEG) encoders instead of the default `av1` decoder. Starting from [6d723c4](https://github.com/huggingface/lerobot/commit/6d723c45a993d1e030dfda20d48a893a74a4fe11), we decided to restrict the available encoders to the three tested in our [video benchmark](https://github.com/huggingface/lerobot/tree/main/benchmarks/video). You may have a look at the corresponding PR #983 to get more details.

That being said, if you think that `MJPEG` could be a good alternative to `av1` feel free to detail why here !

Best,

Caroline.

milong26, 2025-05-11:
> Hi [@milong26](https://github.com/milong26) !
>
> If I understand your issue correctly, you are trying to use the [`MJPEG`](https://en.wikipedia.org/wiki/Motion_JPEG) encoders instead of the default [...]

Yeah your understanding right. I know that I have to use default av1 decoder if I want to follow video benchmarl :D thanks for your answer.

To me I can just use only these two : `YUYV` and `MJPEG` under camera setting. like this
```
v4l2-ctl -d /dev/video0 --list-formats
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'MJPG' (Motion-JPEG, compressed)
        [1]: 'YUYV' (YUYV 4:2:2)
```
but when I use default `YUYV` bandwidth is a problem(cursed). So I have a try on `MJPG` it woks.
I don't know clearly the real differece between these two format(just mjpeg takes smaller space and gets compressed picture) Will `MJPEG` format make much difference to the result?

milong26, 2025-05-12:
Thanks! I will take `MJPEG`  format as a test, after this I'll tell u how about it going. :)

closed 2025-05-12 by milong26

vkamenski, 2025-07-28:
Just adding a data point: I’m using a Logitech C920 on Ubuntu 24.04, and I can confirm that the camera only works reliably with the MJPG  format. While YUYV is also listed by v4l2-ctl, it results in severe frame drops or flickering at higher resolutions (e.g., 1280x720). Forcing MJPG via v4l2-ctl or OpenCV's CAP_PROP_FOURCC is the only stable option in my setup.

forgetwhatuwant, 2025-07-29:
Hi @vkamenski @milong26 ,
Please feel free to test the PR #1558 , i just use MJPG as my opencv format to run smolvla policy from data-collecting to trainning and inference.
Best wish!