huggingface/lerobot #1094: does not accept usb-port from configs.py

mukral, 2025-05-10:
Hello,

script
find_motors_bus_port.py
found COM3 and COM4 (windows environment). 

script
python lerobot/scripts.py --port COM4 --brand feetech --model sts3215 --baudrate 1000000 --ID 1
works and the motor moves.

But when I try to calibrate and run
python lerobot/scripts/control_robot.py --robot.type=so101 --robot.cameras='{}' --control.type=calibrate
I receive an error
serial.serialutil.SerialException: could not open port '/dev/tty.usbmodem585A0076891'
which is the default one from original file.

although port in config.py is set to COM4 and COM3:
...
@RobotConfig.register_subclass("so101")
@dataclass
class So101RobotConfig(ManipulatorRobotConfig):
    calibration_dir: str = ".cache/calibration/so101"
    # `max_relative_target` limits the magnitude of the relative positional target vector for safety purposes.
    # Set this to a positive scalar to have the same value for all motors, or a list that is the same length as
    # the number of motors in your follower arms.
    max_relative_target: int | None = None

    leader_arms: dict[str, MotorsBusConfig] = field(
        default_factory=lambda: {
            "main": FeetechMotorsBusConfig(
                port="COM4",
                motors={
                    # name: (index, model)
                    "shoulder_pan": [1, "sts3215"],
                    "shoulder_lift": [2, "sts3215"],
                    "elbow_flex": [3, "sts3215"],
                    "wrist_flex": [4, "sts3215"],
                    "wrist_roll": [5, "sts3215"],
                    "gripper": [6, "sts3215"],
                },
...
Cache also should be deletet , because I've booted the system (several) times.
Any idea?

short other question: does anybody know how to replace the ' " ' in
  --control.arms='["main_follower"]'?
windows seems to have problems with it  and throws an error
The given value='[main_follower]' is not of a valid input for a list type

Thanks a lot,
Markus

jakmilller, 2025-05-11:
I encountered the same issue on a Linux system. Although I'm sure there's alternative solutions, I fixed it by creating my own configuration file `so100_config.yaml` seen below
```python 
robot:
  type: so100
  cameras: {}
  follower_arms:
    main:
      mock: false
      motors:
        elbow_flex: [3, "sts3215"]
        gripper: [6, "sts3215"]
        shoulder_lift: [2, "sts3215"]
        shoulder_pan: [1, "sts3215"]
        wrist_flex: [4, "sts3215"]
        wrist_roll: [5, "sts3215"]
      port: /dev/ttyACM1
      type: feetech  # Added missing type field
  leader_arms:
    main:
      mock: false
      motors:
        elbow_flex: [3, "sts3215"]
        gripper: [6, "sts3215"]
        shoulder_lift: [2, "sts3215"]
        shoulder_pan: [1, "sts3215"]
        wrist_flex: [4, "sts3215"]
        wrist_roll: [5, "sts3215"]
      port: /dev/ttyACM0
      type: feetech  # Added missing type field
control:
  type: calibrate
  arms: ["main_follower"]
```
Try inserting your appropriate port names into this and running `python3 lerobot/scripts/control_robot.py --config so100_config.yaml`. Hopefully this works for you.

jakmilller, 2025-05-11:
If you get the same error when trying to teleoperate, change the last part of the file to 
```python
control:
  type: teleoperate
```
deleting the `arms` parameter, and run the same command as above.

CarolinePascal, 2025-05-14:
Hi @mukral  !

Could you check if the port you want to use is correctly reported in the configuration displayed when the script is launched ? With the unmodified default [`configs.py`](https://github.com/huggingface/lerobot/blob/main/lerobot/common/robot_devices/robots/configs.py) you should see :

```
INFO 2025-05-14 12:02:43 ol_robot.py:409 {'control': {'arms': ['main_follower']},
 'robot': {'calibration_dir': '.cache/calibration/so101',
           'cameras': {},
           'follower_arms': {'main': {'mock': False,
                                      'motors': {'elbow_flex': [3, 'sts3215'],
                                                 'gripper': [6, 'sts3215'],
                                                 'shoulder_lift': [2,
                                                                   'sts3215'],
                                                 'shoulder_pan': [1, 'sts3215'],
                                                 'wrist_flex': [4, 'sts3215'],
                                                 'wrist_roll': [5, 'sts3215']},
                                      'port': '/dev/tty.usbmodem585A0076891'}},
           'gripper_open_degree': None,
           'leader_arms': {'main': {'mock': False,
                                    'motors': {'elbow_flex': [3, 'sts3215'],
                                               'gripper': [6, 'sts3215'],
                                               'shoulder_lift': [2, 'sts3215'],
                                               'shoulder_pan': [1, 'sts3215'],
                                               'wrist_flex': [4, 'sts3215'],
                                               'wrist_roll': [5, 'sts3215']},
                                    'port': '/dev/tty.usbmodem58760431091'}},
           'max_relative_target': None,
           'microphones': {},
           'mock': False}}
```

If your modifications were correctly taken into account, the output should be modified accordingly.

Best,

Caroline.

ZeroHezi, 2025-05-14:
When running control_robot.py, maybe the program defaults to loading modules from the site-packages directory in the virtual environment, rather than from the current working directory. To resolve this, add the following code at the beginning of the control_robot.py script to insert the current working directory at the front of sys.path, ensuring that modules from the current directory are loaded first:

import sys
import os

# Insert the current working directory at the front of sys.path
current_working_directory = '/home/user/lerobot/'
sys.path.insert(0, current_working_directory)

With this change, Python will prioritize loading modules from the current_working_directory over those in site-packages

mukral, 2025-05-18:
I think the last hint djid the trick. I've also changed all "so100" to "so101" in control_robot.py and at least I receive another error which shows that the system tries to connect COM-ports of the configs.py
ConnectionError: Read failed due to communication error on port COM3 for group_key

Thanks for your support
(But I give up, not another lost sunday with headaches)

CarolinePascal, 2025-05-19:
Hi @mukral,

Glad to see that you solved your issue ! To avoid any other issue with the SO-101, make sure you use the last version of the `main` branch, which has been updated accordingly. Also, try to stick with the installation procedure detailed in the dedicated [example](https://github.com/huggingface/lerobot/blob/main/examples/12_use_so101.md) to avoid any module loading issues ;)

I'm closing this issue then.

Best,

Caroline.

closed 2025-05-19 by CarolinePascal

brainwavecoder9, 2025-05-20:
HI @CarolinePascal I just ran into this also. I think the problem is that the instructions say to run `pip install ".[feetech]"` but that will reference the installed packages and not local changes (e.g., the updated USB port). To pull from user edits, they will need to run `pip install _**-e**_ ".[feetech]"`

The quick fix is to re-install in dev mode (assuming you're following the [so-101 instructions](https://github.com/huggingface/lerobot/blob/main/examples/12_use_so101.md) otherwise you will just need to update your extras accordingly): 
```
pip uninstall lerobot -y
pip install -e ".[feetech]"
```

I couldn't re-open this issue so I created #1129

CarolinePascal, 2025-05-20:
Hi @brainwavecoder9 ,

You're completely right ! As the SO-101 documentation mimics the SO-100 it's hard to spot the differences ;) 

Thanks for your help on that matter ! I'm switching to your new issue for further resolution.

Best,

Caroline.