huggingface/lerobot #1095: Cannot eval policy because dataset statistics are missing

DominiquePaul, 2025-05-10:
### System Info

```Shell
- `lerobot` version: 0.1.0
- Platform: macOS-15.0.1-arm64-arm-64bit
- Python version: 3.10.13
- Huggingface_hub version: 0.31.1
- Dataset version: 3.6.0
- Numpy version: 2.2.5
- PyTorch version (GPU?): 2.7.0 (False)
- Cuda version: N/A
- Using GPU in script?: No, MPS
```

### Information

- [ ] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

We recorded our dataset using Phospho and it is saved on Huggingface (https://huggingface.co/datasets/briannnyee/grabbing_v2/tree/main
We then trained our model using phospho and ACT like this:

```
python lerobot/scripts/train.py \
--policy.type=act \
--dataset.repo_id=briannnyee/grabbing_v2 \
--wandb.enable=true \
--steps=50000 \
--output_dir=lerobot/outputs/train/grg_grabbing_v2
```

# Push the model to Hugging Face. 
python lerobot/scripts/push_pretrained.py \
    --pretrained_path=lerobot/outputs/train/grg_grabbing_v2/checkpoints/last/pretrained_model \
    --repo_id=dopaul/zrh_grasping_v2_small

The ultimate goal: Call a single policy execution from an LLM using tool use
The main thing we want to do is to have an LLM invoke an action on the robot. For this we will need a wrapper that calls one of the policies we have trained. 

LeRobot has examples for full evaluation, but no example, how to just execute a single policy once. We want to implement this ourself using a server running on a GPU on the cloud and a client sending API requests, as running the policy on our macbook would be too slow. 

The part we're stuck at: Executing the eval script from the readme
In order to figure this out we first wanted to just run the evaluation script provided here: https://github.com/huggingface/lerobot/blob/a445d9c9da6bea99a8972daa4fe1fdd053d711d2/examples/10_use_so100.md#k-evaluate-your-policy
When we run the script, replacing the model name for ours like this...

```
python lerobot/scripts/control_robot.py \
  --robot.type=so100 \
  --control.type=record \
  --control.fps=30 \
  --control.single_task="Grasp a lego block and put it in the bin." \
  --control.repo_id=dopaul/eval_zrh_grasping_v2_small \
  --control.tags='["tutorial"]' \
  --control.warmup_time_s=5 \
  --control.episode_time_s=30 \
  --control.reset_time_s=30 \
  --control.num_episodes=10 \
  --control.push_to_hub=true \
  --control.policy.path=dopaul/zrh_grasping_v2_small
```

...this unfortunately results in the following error message:

```
WARNING:root:Device 'cuda' is not available. Switching to 'mps'.
Connecting main follower arm.
Connecting main leader arm.
Activating torque on main follower arm.
Traceback (most recent call last):
  File "/Users/dpaul/Documents/09 Robotics/lerobot/lerobot/scripts/control_robot.py", line 437, in <module>
    control_robot()
  File "/Users/dpaul/Documents/09 Robotics/lerobot/lerobot/configs/parser.py", line 227, in wrapper_inner
    response = fn(cfg, *args, **kwargs)
  File "/Users/dpaul/Documents/09 Robotics/lerobot/lerobot/scripts/control_robot.py", line 421, in control_robot
    record(robot, cfg.control)
  File "/Users/dpaul/Documents/09 Robotics/lerobot/lerobot/common/robot_devices/utils.py", line 42, in wrapper
    raise e
  File "/Users/dpaul/Documents/09 Robotics/lerobot/lerobot/common/robot_devices/utils.py", line 38, in wrapper
    return func(robot, *args, **kwargs)
  File "/Users/dpaul/Documents/09 Robotics/lerobot/lerobot/scripts/control_robot.py", line 299, in record
    record_episode(
  File "/Users/dpaul/Documents/09 Robotics/lerobot/lerobot/common/robot_devices/control_utils.py", line 200, in record_episode
    control_loop(
  File "/Users/dpaul/Documents/09 Robotics/lerobot/lerobot/common/datasets/image_writer.py", line 36, in wrapper
    raise e
  File "/Users/dpaul/Documents/09 Robotics/lerobot/lerobot/common/datasets/image_writer.py", line 29, in wrapper
    return func(*args, **kwargs)
  File "/Users/dpaul/Documents/09 Robotics/lerobot/lerobot/common/robot_devices/control_utils.py", line 255, in control_loop
    pred_action = predict_action(
  File "/Users/dpaul/Documents/09 Robotics/lerobot/lerobot/common/robot_devices/control_utils.py", line 120, in predict_action
    action = policy.select_action(observation)
  File "/opt/homebrew/anaconda3/envs/lerobot/lib/python3.10/site-packages/torch/utils/_contextlib.py", line 116, in decorate_context
    return func(*args, **kwargs)
  File "/Users/dpaul/Documents/09 Robotics/lerobot/lerobot/common/policies/act/modeling_act.py", line 119, in select_action
    batch = self.normalize_inputs(batch)
  File "/opt/homebrew/anaconda3/envs/lerobot/lib/python3.10/site-packages/torch/nn/modules/module.py", line 1751, in _wrapped_call_impl
    return self._call_impl(*args, **kwargs)
  File "/opt/homebrew/anaconda3/envs/lerobot/lib/python3.10/site-packages/torch/nn/modules/module.py", line 1762, in _call_impl
    return forward_call(*args, **kwargs)
  File "/opt/homebrew/anaconda3/envs/lerobot/lib/python3.10/site-packages/torch/utils/_contextlib.py", line 116, in decorate_context
    return func(*args, **kwargs)
  File "/Users/dpaul/Documents/09 Robotics/lerobot/lerobot/common/policies/normalize.py", line 169, in forward
    assert not torch.isinf(mean).any(), _no_stats_error_str("mean")
AssertionError: `mean` is infinity. You should either initialize with `stats` as an argument, or use a pretrained model.
```

I am not sure where we could be getting these statistics. We are just copying all the commands from the readme. We've tried debugging this with LLMs and reading the code, but we're very stuck here and not sure what the cause of this is/how we could fix it.

Thank you,
Dominique + Team

### Expected behavior

We'd expect the policy to run.

DominiquePaul, 2025-05-10:
Ok, I figured this out eventually: 

My camera's in `lerobot/common/robot_devices/robots/configs.py` were named differently than in `MY_HF_DATASET/meta/stats.json` 👍🏼

CarolinePascal, 2025-05-14:
Hi @DominiquePaul !

I'm closing this issue as it seems you solved it on your own ;)

closed 2025-05-14 by CarolinePascal