huggingface/lerobot #1101: ValueError: No integer found between bounds [low_factor=np.float32(-0.001953125), upp_factor=np.float32(-0.001953125)]

qingx-cyber, 2025-05-13:
### System Info

```Shell
2025,ubantu,python3.10. when doing teleoperation
```

### Information

- [ ] One of the scripts in the examples/ folder of LeRobot
- [x] My own task or dataset (give details below)

### Reproduction

python lerobot/scripts/control_robot.py   --robot.type=so100   --robot.cameras='{}'   --control.type=teleoperate 

### Expected behavior

How to deal with it.

SangbumChoi, 2025-05-20:
https://github.com/huggingface/lerobot/issues/1098#issue-3058106674

SangbumChoi, 2025-05-20:
I also faced this issue but removing cache did not helped @qingx-cyber

SangbumChoi, 2025-05-21:
I have solved the problem. For me the reason was not setting the leader arm calibration