huggingface/lerobot #1105: `type` field missing from saved policy

ankile, 2025-05-13:
### System Info

```Shell
- `lerobot` version: 0.1.0
- Platform: Linux-6.8.0-58-generic-x86_64-with-glibc2.35
- Python version: 3.10.15
- Huggingface_hub version: 0.30.2
- Dataset version: 3.6.0
- Numpy version: 1.26.0
- PyTorch version (GPU?): 2.7.0+cu126 (True)
- Cuda version: 12060
- Using GPU in script?: Yes, Nvidia GeForce RTX 4090
```

### Information

- [x] One of the scripts in the examples/ folder of LeRobot
- [x] My own task or dataset (give details below)

### Reproduction

Hi, I'm  getting this issue when trying to load a diffusion policy that was created with the "official" training script: 

python lerobot/scripts/train.py \                                          
    --dataset.repo_id=ankile/franka-lift \
    --policy.type=diffusion \             
    --wandb.enable=true

When calling policy = DiffusionPolicy.from_pretrained(pretrained_policy_path) with the resulting saved models, I get

draccus.utils.ParsingError: Expected a dict with a 'type' key for <class 'lerobot.configs.policies.PreTrainedConfig'>, got {'n_obs_steps': 2,....

which seems strange to me because the repo is freshly installed as of a couple of days ago and using the standard training script and standard DP config.

### Expected behavior

I would expect calling `DiffusionPolicy.from_pretrained(pretrained_policy_path)` with the output of the "official" `train.py` script to work without having to edit the `config.json` to add the `type: diffusion` key.

xiaojie, 2025-05-22:
the same bug: type field missing from saved robot config

Shunying, 2025-05-23:
same bug for pi0fast

liuhuxian, 2025-05-27:
This bug is caused by an incorrect version of draccus.
I fixed it by the following command, you can try it.
```
pip uninstall draccus
pip install draccus==0.10.0
```

Daniel891116, 2025-06-03:
> This bug is caused by an incorrect version of draccus. I fixed it by the following command, you can try it.
> 
> ```
> pip uninstall draccus
> pip install draccus==0.10.0
> ```

I solve the problem with this fix. Thank you.

closed 2025-12-16 by github-actions[bot]