huggingface/lerobot #1109: Tag or releases of lerobot?

traversaro, 2025-05-14:
Thanks a lot for all the great work on lerobot! 

I was wondering if you plan to eventually either tag or release the software in this repo? While it is already possible to install in a reproducible way a specific version of lerobot via:

~~~
pip install "lerobot @ git+https://github.com/huggingface/lerobot.git@a445d9c9da6bea99a8972daa4fe1fdd053d711d2"
~~~

having releases would be convenient for the following reasons:

* It is easier to express compatibility bounds (i.e. I know that my software is compatuble with lerobot >= 0.1.2 and lerobot < 0.2.5, while expressing this with raw commits is tricky)
* It can be used to upload versions to pip (for example, as of May 2025 `pip install lerobot` works fine, but it install a really old versions of lerobot). This may permit for example to upload to PyPI software that depends on recent features of lerobot.

I briefly searched if there were already discussion on this but I could not find any issue or discussion related to that, if there is something that I overlooked, feel free to let me know, thanks.

traversaro, 2025-05-14:
Just to clarify, note that even if there is no strict back-compatibility policy, the aforementioned advantages of tags and release still hold. I just wanted to point this out, as it is common to conflate "doing tag/releases" with "doing a tag and doing long-term mantainance of that version", while this issue was only asking about doing tags, not about anything else.

traversaro, 2025-05-14:
> I briefly searched if there were already discussion on this but I could not find any issue or discussion related to that

Probably https://github.com/Physical-Intelligence/openpi/issues/313#issuecomment-2672104794 is related.

imstevenpmwork, 2025-06-19:
This is included in the near future roadmap 😄

traversaro, 2025-08-04:
This was solved:
* https://github.com/huggingface/lerobot/releases/tag/v0.3.2
* https://pypi.org/project/lerobot/#history

Thanks a lot to everyone that worked on this!

fyi @S-Dafarra @giotherobot @mtiezzi @apicis @carloscp3009

closed 2025-10-17 by imstevenpmwork