huggingface/lerobot #1110: Error occurred during motor configuration: No motors detected. Please ensure you have one motor connected.

SangbumChoi, 2025-05-15:
https://github.com/huggingface/lerobot/blob/main/examples/12_use_so101.md#set-motor-ids

I am stuck with this part. What am I missing?

```
(lerobot) sangbumchoi@sangbumui-MacBookAir lerobot % python lerobot/scripts/configure_motor.py \
  --port /dev/tty.usbmodem5A460825151 \
  --brand feetech \
  --model sts3215 \
  --baudrate 1000000 \
  --ID 1
Connected on port /dev/tty.usbmodem5A460825151
Scanning all baudrates and motor indices
100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 9/9 [00:00<00:00, 14.61it/s]
Setting bus baud rate to 128000. Previously 1000000.
100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 9/9 [00:00<00:00, 14.43it/s]
Setting bus baud rate to 500000. Previously 128000.
100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 9/9 [00:00<00:00, 14.60it/s]
Setting bus baud rate to 115200. Previously 500000.
100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 9/9 [00:00<00:00, 14.33it/s]
Setting bus baud rate to 57600. Previously 115200.
100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 9/9 [00:00<00:00, 14.14it/s]
Setting bus baud rate to 38400. Previously 57600.
100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 9/9 [00:00<00:00, 13.92it/s]
Setting bus baud rate to 19200. Previously 38400.
100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 9/9 [00:00<00:00, 13.25it/s]
Setting bus baud rate to 250000. Previously 19200.
100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 9/9 [00:00<00:00, 14.54it/s]
Error occurred during motor configuration: No motors detected. Please ensure you have one motor connected.
Disconnected from motor bus.
```

[image]

gilspringer, 2025-05-15:
same, I did the FOLLOWER yesterday, everything was good, and today I try to do the LEADER but I have the same error as you, I am also on mac.

gilspringer, 2025-05-15:
I see you don't have the jumpers on USB

SangbumChoi, 2025-05-15:
@gilspringer how should I connect the jumpers (I think you mean the yellow thing) A or B?

gilspringer, 2025-05-16:
[image]

Yes like that

SangbumChoi, 2025-05-16:
@gilspringer connecting jumpers did not solve the problem :(

gilspringer, 2025-05-16:
@SangbumChoi I think the problem comes from the USB hub, I just succeeded on my side by connecting directly to my Mac and it works now :)

SangbumChoi, 2025-05-16:
@SangbumChoi Mine works to :)

ConnectionError: Read failed due to communication error on port /dev/tty.usbmodem5A460825151 for group_key Torque_Enable_shoulder_pan_shoulder_lift_elbow_flex_wrist_flex_wrist_roll_gripper: [TxRxResult] Incorrect status packet!

This happens the next

closed 2025-05-16 by SangbumChoi