huggingface/lerobot #1111: Unrecognized argument policy.path. How to load a pretrained model?

milong26, 2025-05-15:
When I run this command:
```
python lerobot/scripts/control_robot.py --robot.type so100 --control.type record --control.fps 30 --control.single_task "Grasp a yellow tape and put it to yellow square." --control.repo_id a_cam_1/result --control.tags '["tutorial"]' --control.warmup_time_s 5 --control.episode_time_s 30 --control.reset_time_s 10 --control.m_episodes 1 --control.push_to_hub false --control.policy,path output/checkpoints/last/pretrained_model
```

I got:
```
usage: control_robot.py [-h] [--config_path str] [--robot str] [--robot.type {aloha,koch,koch_bimanual,moss,so101,so100,stretch,lekiwi}] [--robot.gripper_open_degree str]
                        [--robot.max_relative_target str] [--robot.ip str] [--robot.port str] [--robot.video_port str] [--robot.cameras str] [--robot.calibration_dir str]
                        [--robot.leader_arms str] [--robot.follower_arms str] [--robot.teleop_keys str] [--robot.mock str] [--control str]
                        [--control.type {calibrate,teleoperate,record,replay,remote_robot}] [--control.arms str] [--control.teleop_time_s str] [--control.single_task str]
                        [--policy str] [--control.policy.type {act,diffusion,pi0,tdmpc,vqbet,pi0fast}] [--control.policy.replace_final_stride_with_dilation str]
                        [--control.policy.pre_norm str] [--control.policy.dim_model str] [--control.policy.n_heads str] [--control.policy.dim_feedforward str]
                        [--control.policy.feedforward_activation str] [--control.policy.n_encoder_layers str] [--control.policy.n_decoder_layers str]
                        [--control.policy.use_vae str] [--control.policy.n_vae_encoder_layers str] [--control.policy.temporal_ensemble_coeff str]
                        [--control.policy.kl_weight str] [--control.policy.optimizer_lr_backbone str] [--control.policy.drop_n_last_frames str]
                        [--control.policy.use_separate_rgb_encoder_per_camera str] [--control.policy.down_dims str] [--control.policy.kernel_size str]
                        [--control.policy.n_groups str] [--control.policy.diffusion_step_embed_dim str] [--control.policy.use_film_scale_modulation str]
                        [--control.policy.noise_scheduler_type str] [--control.policy.num_train_timesteps str] [--control.policy.beta_schedule str]
                        [--control.policy.beta_start str] [--control.policy.beta_end str] [--control.policy.prediction_type str] [--control.policy.clip_sample str]
                        [--control.policy.clip_sample_range str] [--control.policy.num_inference_steps str] [--control.policy.do_mask_loss_for_padding str]
                        [--control.policy.scheduler_name str] [--control.policy.num_steps str] [--control.policy.attention_implementation str]
                        [--control.policy.train_expert_only str] [--control.policy.train_state_proj str] [--control.policy.n_action_repeats str] [--control.policy.horizon str]
                        [--control.policy.image_encoder_hidden_dim str] [--control.policy.state_encoder_hidden_dim str] [--control.policy.latent_dim str]
                        [--control.policy.q_ensemble_size str] [--control.policy.mlp_dim str] [--control.policy.discount str] [--control.policy.use_mpc str]
                        [--control.policy.cem_iterations str] [--control.policy.max_std str] [--control.policy.min_std str] [--control.policy.n_gaussian_samples str]
                        [--control.policy.n_pi_samples str] [--control.policy.uncertainty_regularizer_coeff str] [--control.policy.n_elites str]
                        [--control.policy.elite_weighting_temperature str] [--control.policy.gaussian_mean_momentum str] [--control.policy.max_random_shift_ratio str]
                        [--control.policy.reward_coeff str] [--control.policy.expectile_weight str] [--control.policy.value_coeff str] [--control.policy.consistency_coeff str]
                        [--control.policy.advantage_scaling str] [--control.policy.pi_coeff str] [--control.policy.temporal_decay_coeff str]
                        [--control.policy.target_model_momentum str] [--control.policy.n_action_pred_token str] [--control.policy.action_chunk_size str]
                        [--control.policy.vision_backbone str] [--control.policy.crop_shape str] [--control.policy.crop_is_random str]
                        [--control.policy.pretrained_backbone_weights str] [--control.policy.use_group_norm str] [--control.policy.spatial_softmax_num_keypoints str]
                        [--control.policy.n_vqvae_training_steps str] [--control.policy.vqvae_n_embed str] [--control.policy.vqvae_embedding_dim str]
                        [--control.policy.vqvae_enc_hidden_dim str] [--control.policy.gpt_block_size str] [--control.policy.gpt_input_dim str]
                        [--control.policy.gpt_output_dim str] [--control.policy.gpt_n_layer str] [--control.policy.gpt_n_head str] [--control.policy.gpt_hidden_dim str]
                        [--control.policy.dropout str] [--control.policy.mlp_hidden_dim str] [--control.policy.offset_loss_weight str]
                        [--control.policy.primary_code_loss_weight str] [--control.policy.secondary_code_loss_weight str] [--control.policy.bet_softmax_temperature str]
                        [--control.policy.sequentially_select str] [--control.policy.optimizer_vqvae_lr str] [--control.policy.optimizer_vqvae_weight_decay str]
                        [--control.policy.n_obs_steps str] [--control.policy.normalization_mapping str] [--control.policy.input_features str]
                        [--control.policy.output_features str] [--control.policy.device str] [--control.policy.use_amp str] [--control.policy.chunk_size str]
                        [--control.policy.n_action_steps str] [--control.policy.max_state_dim str] [--control.policy.max_action_dim str]
                        [--control.policy.resize_imgs_with_padding str] [--control.policy.interpolate_like_pi str] [--control.policy.empty_cameras str]
                        [--control.policy.adapt_to_pi_aloha str] [--control.policy.use_delta_joint_actions_aloha str] [--control.policy.tokenizer_max_length str]
                        [--control.policy.proj_width str] [--control.policy.max_decoding_steps str] [--control.policy.fast_skip_tokens str]
                        [--control.policy.max_input_seq_len str] [--control.policy.use_cache str] [--control.policy.freeze_vision_encoder str]
                        [--control.policy.freeze_lm_head str] [--control.policy.optimizer_lr str] [--control.policy.optimizer_betas str] [--control.policy.optimizer_eps str]
                        [--control.policy.optimizer_weight_decay str] [--control.policy.scheduler_warmup_steps str] [--control.policy.scheduler_decay_steps str]
                        [--control.policy.scheduler_decay_lr str] [--control.policy.checkpoint_path str] [--control.policy.padding_side str] [--control.policy.precision str]
                        [--control.policy.grad_clip_norm str] [--control.policy.relaxed_action_decoding str] [--control.warmup_time_s str] [--control.episode_time_s str]
                        [--control.reset_time_s str] [--control.num_episodes str] [--control.video str] [--control.push_to_hub str] [--control.private str] [--control.tags str]
                        [--control.num_image_writer_processes str] [--control.num_image_writer_threads_per_camera str] [--control.resume str] [--control.repo_id str]
                        [--control.episode str] [--control.root str] [--control.fps str] [--control.play_sounds str] [--control.log_interval str] [--control.display_data str]
                        [--control.viewer_ip str] [--control.viewer_port str]
control_robot.py: error: unrecognized arguments: --control.policy.path output/checkpoints/last/pretrained_model
```
lerobot version:42bf1e8b9df3d38b3898b24e46b5e0386910c466

milong26, 2025-05-15:
I tried this:
```
--robot.type so100 --control.type record --control.fps 30 --control.single_task "Grasp a yellow tape and put it to yellow square." --control.repo_id a_cam_1/eval_result --control.tags '["tutorial"]' --control.warmup_time_s 5 --control.episode_time_s 30 --control.reset_time_s 10 --control.num_episodes 1 --control.push_to_hub false --control.policy.type act --control.policy.pretrained_path outputs/train/act_so100_test/checkpoints/last/pretrained_model
```
When I just use `--control.policy.type act ` it's ok but no pretrained model path in make_policy() function. 
```python
    if cfg.pretrained_path:
        print("path=",cfg.pretrained_path)
        # Load a pretrained policy and override the config if needed (for example, if there are inference-time
        # hyperparameters that we want to vary).
        kwargs["pretrained_name_or_path"] = cfg.pretrained_path
        policy = policy_cls.from_pretrained(**kwargs)
```

But when I add `--control.policy.pretrained_path outputs/train/act_so100_test/checkpoints/last/pretrained_model` It raises error in lerobot/config/parser.py:

```python
else:
                    cfg = draccus.parse(config_class=argtype, config_path=config_path, args=cli_args)
```
reporting this:
Exception has occurred: SystemExit
2
  File "/xx/lerobot/lerobot/configs/parser.py", line 226, in wrapper_inner
    cfg = draccus.parse(config_class=argtype, config_path=config_path, args=cli_args)
  File "/xx/lerobot/lerobot/scripts/control_robot.py", line 437, in <module>
    control_robot()
SystemExit: 2

milong26, 2025-05-15:
I can't register cfg,policy.pretrained_path=/outputs/train/act_so100_test/checkpoints/last/pretrained_model in `control_robot.py`  How to do it?

milong26, 2025-05-15:
Ok I succeeded it by modifying many.

CarolinePascal, 2025-05-19:
Hi @milong26 !

The following one-line command works on my side :
```bash
python lerobot/scripts/control_robot.py --robot.type=so100 --control.type=record --control.fps=30 --control.single_task="Grasp a yellow tape and put it to yellow square." --control.repo_id=a_cam_1/eval_result --control.tags='["tutorial"]' --control.warmup_time_s=5 --control.episode_time_s=30 --control.reset_time_s=10 --control.num_episodes=1 --control.push_to_hub=false --control.policy.path=outputs/train/act_so100_test/checkpoints/last/pretrained_model
```

Can you confirm it also works on your side ? Also please make sure all provided paths are correct (`--control.policy.path` and `--control.repo_id`).

Best,

Caroline.

milong26, 2025-05-19:
> Hi [@milong26](https://github.com/milong26) !
> 
> The following one-line command works on my side :
> 
> python lerobot/scripts/control_robot.py --robot.type=so100 --control.type=record --control.fps=30 --control.single_task="Grasp a yellow tape and put it to yellow square." --control.repo_id=a_cam_1/eval_result --control.tags='["tutorial"]' --control.warmup_time_s=5 --control.episode_time_s=30 --control.reset_time_s=10 --control.num_episodes=1 --control.push_to_hub=false --control.policy.path=outputs/train/act_so100_test/checkpoints/last/pretrained_model
> Can you confirm it also works on your side ? Also please make sure all provided paths are correct (`--control.policy.path` and `--control.repo_id`).
> 
> Best,
> 
> Caroline.

Maybe I used the wrong version. I'll check it,thanks!

milong26, 2025-06-24:
I met it again even in latest version(2b71789e15c35418b1ccecbceb81f4a598bfd883) when I try to debug it.
When I debug with launch.json
```json
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [

        {
            "name": "Python Debugger: Current File with Arguments",
            "type": "debugpy",
            "request": "launch",
            "python": "conda_path/anaconda3/envs/lerobot/bin/python",
            "program": "lerobot_path/lerobot/scripts/train.py",
            "console": "integratedTerminal",
            "cwd": "working_directory",
            "env": {
                "PYTHONPATH": "working_directory:${env:PYTHONPATH}" // 关键设置
              },
            "args": [
                "--policy.path", "I downloaded models here/models/forsmolvla/smolvla_base",
                "--dataset.repo_id", "lerobot/svla_so101_pickplace", # just for decorating XD
                "--batch_size", "64",
                "--steps", "20000",
                "--output_dir", "outputs/train/my_smolvla",
                "--job_name", "my_smolvla_training",
                "--policy.device", "cuda",
                "--wandb.enable", "false",
                "--dataset.root", "my local fine-tune dataset /training_data/622work"
            ]
        }
    ]
}
```
Then it stopped at ` /lerobot/configs/parser.py` 
```python
def wrap(config_path: Path | None = None):
    """
    HACK: Similar to draccus.wrap but does three additional things:
        - Will remove '.path' arguments from CLI in order to process them later on.
        - If a 'config_path' is passed and the main config class has a 'from_pretrained' method, will
          initialize it from there to allow to fetch configs from the hub directly
        - Will load plugins specified in the CLI arguments. These plugins will typically register
            their own subclasses of config classes, so that draccus can find the right class to instantiate
            from the CLI '.type' arguments
    """

    def wrapper_outer(fn):
        @wraps(fn)
        def wrapper_inner(*args, **kwargs):
            argspec = inspect.getfullargspec(fn)
            argtype = argspec.annotations[argspec.args[0]]
            if len(args) > 0 and type(args[0]) is argtype:
                cfg = args[0]
                args = args[1:]
            else:
                cli_args = sys.argv[1:]
                plugin_args = parse_plugin_args(PLUGIN_DISCOVERY_SUFFIX, cli_args)
                for plugin_cli_arg, plugin_path in plugin_args.items():
                    try:
                        load_plugin(plugin_path)
                    except PluginLoadError as e:
                        # add the relevant CLI arg to the error message
                        raise PluginLoadError(f"{e}\nFailed plugin CLI Arg: {plugin_cli_arg}") from e
                    cli_args = filter_arg(plugin_cli_arg, cli_args)
                config_path_cli = parse_arg("config_path", cli_args)
                if has_method(argtype, "__get_path_fields__"):
                    path_fields = argtype.__get_path_fields__()
[... 121 lines cut ...]
                [--policy.actor_learner_config.policy_parameters_push_frequency str]
                [--policy.actor_learner_config.queue_get_timeout str] [--concurrency str]
                [--policy.concurrency.actor str] [--policy.concurrency.learner str]
                [--policy.use_torch_compile str] [--policy.n_obs_steps str] [--policy.normalization_mapping str]
                [--policy.input_features str] [--policy.output_features str] [--policy.device str]
                [--policy.use_amp str] [--policy.name str] [--policy.num_classes str] [--policy.hidden_dim str]
                [--policy.latent_dim str] [--policy.image_embedding_pooling_dim str] [--policy.dropout_rate str]
                [--policy.model_name str] [--policy.model_type str] [--policy.num_cameras str]
                [--policy.learning_rate str] [--policy.weight_decay str] [--policy.grad_clip_norm str]
                [--output_dir str] [--job_name str] [--resume str] [--seed str] [--num_workers str]
                [--batch_size str] [--steps str] [--eval_freq str] [--log_freq str] [--save_checkpoint str]
                [--save_freq str] [--use_policy_training_preset str] [--optimizer str]
                [--optimizer.type {adam,adamw,sgd,multi_adam}] [--optimizer.betas str] [--optimizer.eps str]
                [--optimizer.momentum str] [--optimizer.dampening str] [--optimizer.nesterov str]
                [--optimizer.lr str] [--optimizer.weight_decay str] [--optimizer.grad_clip_norm str]
                [--optimizer.optimizer_groups str] [--scheduler str]
                [--scheduler.type {diffuser,vqbet,cosine_decay_with_warmup}] [--scheduler.name str]
                [--scheduler.num_vqvae_training_steps str] [--scheduler.num_cycles str]
                [--scheduler.num_warmup_steps str] [--scheduler.num_decay_steps str] [--scheduler.peak_lr str]
                [--scheduler.decay_lr str] [--eval str] [--eval.n_episodes str] [--eval.batch_size str]
                [--eval.use_async_envs str] [--wandb str] [--wandb.enable str] [--wandb.disable_artifact str]
                [--wandb.project str] [--wandb.entity str] [--wandb.notes str] [--wandb.run_id str]
                [--wandb.mode str]
train.py: error: unrecognized arguments: --policy.path /data/zly/junwork/lerobot_diy/models/forsmolvla/smolvla_base
```

milong26, 2025-06-24:
:P I don't know why, when I change args in json to: = it works well,like this:
```json

            "args": [
                "--policy.path=I downloaded models here/models/forsmolvla/smolvla_base",
                "--dataset.repo_id=lerobot/svla_so101_pickplace", # just for decorating XD
                "--batch_size=64",
                "--steps=20000",
                "--output_dir=outputs/train/my_smolvla",
                "--job_name=my_smolvla_training",
                "--policy.device=cuda",
                "--wandb.enable"=false",
                "--dataset.root=my local fine-tune dataset /training_data/622work"
            ]
```
In my imagine, `=` is same as ` ` but it seems not here. Maybe it's about parse?