huggingface/lerobot #1121: [TxRxResult] Incorrect status packet! #1010

SangbumChoi, 2025-05-17:
Traceback (most recent call last):
  File "/Users/sangbumchoi/Downloads/lerobot/lerobot/scripts/control_robot.py", line 437, in <module>
    control_robot()
  File "/Users/sangbumchoi/Downloads/lerobot/lerobot/configs/parser.py", line 227, in wrapper_inner
    response = fn(cfg, *args, **kwargs)
  File "/Users/sangbumchoi/Downloads/lerobot/lerobot/scripts/control_robot.py", line 415, in control_robot
    calibrate(robot, cfg.control)
  File "/Users/sangbumchoi/Downloads/lerobot/lerobot/common/robot_devices/utils.py", line 42, in wrapper
    raise e
  File "/Users/sangbumchoi/Downloads/lerobot/lerobot/common/robot_devices/utils.py", line 38, in wrapper
    return func(robot, *args, **kwargs)
  File "/Users/sangbumchoi/Downloads/lerobot/lerobot/scripts/control_robot.py", line 228, in calibrate
    robot.connect()
  File "/Users/sangbumchoi/Downloads/lerobot/lerobot/common/robot_devices/robots/manipulator.py", line 256, in connect
    self.activate_calibration()
  File "/Users/sangbumchoi/Downloads/lerobot/lerobot/common/robot_devices/robots/manipulator.py", line 331, in activate_calibration
    calibration = load_or_run_calibration_(name, arm, "follower")
  File "/Users/sangbumchoi/Downloads/lerobot/lerobot/common/robot_devices/robots/manipulator.py", line 321, in load_or_run_calibration_
    calibration = run_arm_manual_calibration(arm, self.robot_type, name, arm_type)
  File "/Users/sangbumchoi/Downloads/lerobot/lerobot/common/robot_devices/robots/feetech_calibration.py", line 443, in run_arm_manual_calibration
    if (arm.read("Torque_Enable") != TorqueMode.DISABLED.value).any():
  File "/Users/sangbumchoi/Downloads/lerobot/lerobot/common/robot_devices/motors/feetech.py", line 754, in read
    raise ConnectionError(
ConnectionError: Read failed due to communication error on port /dev/tty.usbmodem5A460825151 for group_key Torque_Enable_shoulder_pan_shoulder_lift_elbow_flex_wrist_flex_wrist_roll_gripper: [TxRxResult] Incorrect status packet!

SangbumChoi, 2025-05-17:
https://github.com/huggingface/lerobot/issues/1010#issue-3008821872

CarolinePascal, 2025-05-19:
Hi @SangbumChoi !

I'm closing this issue so that we can focus on the older duplicate #1010 .

Best,

Caroline.

closed 2025-05-20 by SangbumChoi