huggingface/lerobot #1126: visualize_dataset_html.py can't show videos

Hpeox, 2025-05-19:
### System Info

```Shell
- `lerobot` version: 0.1.0
- Platform: Windows-10-10.0.22631-SP0 #i thought it should be Win11?
- Python version: 3.10.16
- Huggingface_hub version: 0.29.1
- Dataset version: 3.3.2
- Numpy version: 2.1.3
- PyTorch version (GPU?): 2.6.0+cu126 (True)
- Cuda version: 12060
- Using GPU in script?: True
```

### Information

- [ ] One of the scripts in the examples/ folder of LeRobot
- [x] My own task or dataset (give details below)

### Reproduction

I tried to visualize my local dataset through
```
python lerobot/scripts/visualize_dataset_html.py --root= --repo-id 
```

the videos just can't load. 

right click to the video and copy the link find the link just like: "/static/videos%5Cchunk-000%5Cobservation.images.wrist%5Cepisode_000001.mp4" with url decoded character "%5C". If I replace this with "/" the video can be played.

I think maybe it's the problem from the broswer (edge)

### Expected behavior

if I change some code like this in visualize_dataset_html.py everything will be ok:

```
videos_info_tmp = [
                {"url": url_for("static", filename=video_path), "filename": video_path.parent.name}
                for video_path in video_paths
            ]
videos_info = [{"url": video_info['url'].replace("%5C","/"),"filename":video_info["filename"]} for video_info in videos_info_tmp]
```
in [line 176](https://github.com/huggingface/lerobot/blob/a445d9c9da6bea99a8972daa4fe1fdd053d711d2/lerobot/scripts/visualize_dataset_html.py#L176C13-L176C25)

CarolinePascal, 2025-05-19:
Hi @Hpeox !

Have you tried loading the videos using another web browser ? Usually Chrome, Firefox and Safari do the trick ;)

Best,

Caroline.

Hpeox, 2025-05-19:
> Hi [@Hpeox](https://github.com/Hpeox) !
> 
> Have you tried loading the videos using another web browser ? Usually Chrome, Firefox and Safari do the trick ;)
> 
> Best,
> 
> Caroline.

@CarolinePascal  I‘ve tried on firefox, the browser get a video link like "/static/videos\chunk-000\observation.images.wrist\episode_000000.mp4", which still can't be played. Manually replace the '\' with '/' will fix this.

CarolinePascal, 2025-05-19:
OK, I just noticed that the script [`visualize_dataset_html.py`](https://github.com/huggingface/lerobot/blob/main/lerobot/scripts/visualize_dataset_html.py) has not been updated to use our new [visualizer](https://huggingface.co/spaces/lerobot/visualize_dataset). The update will hopefully solve your issue :) 

@mishig25 Is there currently a way to run your new visualizer locally ? Or have we to upload the dataset on the hub to visualize it online ?

Hpeox, 2025-05-19:
> OK, I just noticed that the script [`visualize_dataset_html.py`](https://github.com/huggingface/lerobot/blob/main/lerobot/scripts/visualize_dataset_html.py) has not been updated to use our new [...]

thanks, I think I am looking forward to its offline version