huggingface/lerobot #1129: Instructions should use editable mode (-e) to prevent configuration inconsistencies

brainwavecoder9, 2025-05-20:
### System Info

```Shell
## Issue Description
When installing the package using the [current instructions](https://github.com/huggingface/lerobot/blob/main/examples/12_use_so101.md) (`cd lerobot && pip install ".[feetech]"`), local modifications to configuration files are ignored during runtime. This is because the package is installed in standard mode, which copies files to site-packages instead of referencing the working directory.

## Steps to repro and & Expected behavior below

## Actual Behavior
The system ignores local changes and uses the default configurations from the installed package.

## Solution
Update installation instructions to recommend editable mode for development:

Instead of `pip install ".[feetech]"` 
Switch to `pip install -e ".[feetech]"`

I'm happy to submit the PR but wanted to confirm expected behavior with @CarolinePascal
```

### Information

- [x] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

## Steps to Reproduce
Note that this is a duplicate of #1094 but I don't have permission to re-open the issue.  

1. Modify configuration files (e.g., to update port settings in `lerobot/common/robot_devices/robots/configs.py`)
2. Install using `pip install ".[feetech]"` per current instructions
3. Run `python lerobot/scripts/control_robot.py --robot.type=so101 --control.type=calibrate`
4. Observe error using default port: `serial.serialutil.SerialException: could not open port '/dev/tty.usbmodem585A0076891'` despite updating the configuration

### Expected behavior

## Expected Behavior
I would think that the local configuration changes are expected to be reflected when running the scripts.

CarolinePascal, 2025-05-20:
Hi @brainwavecoder9 ,

Thanks again for spotting this error :) It also seems to me that restarting the shell is only necessary when installing `conda` for the first time (as it is done in the SO-100 documentation as well).

Could you confirm that both changes are OK for you @pkooij ?

Best,

Caroline.

CarolinePascal, 2025-05-20:
Issue fixed in #1133 !

closed 2025-05-20 by CarolinePascal

brainwavecoder9, 2025-05-20:
🥳