huggingface/lerobot #1130: Drive mode reversed on calibration.

brainwavecoder9, 2025-05-20:
I had an issue where after calibrating drive_mode was reversed for one of my motors (0 vs. 1) as a result, moving the leader in one direction caused the follower to go the opposite direction.

Saw some suggestions that moving it through the full range of motion resolved this but I wasn't able to get that to work. I could also see cases where this could be problematic during initial setup. @Lemin2 suggested to always set this to 0 across the board, which does seem like a good fix, unless there's a reason want to control reverse mode. 

In any case I would expect the calibration process to be consistent for both arms, else this issue will be encountered. If reverse mode is needed maybe have a step in the calibration processes to ensure consistency.

FYI in case anyone encounters this the solution is to go into `.cache/calibration/<arm>/<each of your arms>.json`

Seems to be the same cause for #441 and #930

brainwavecoder9, 2025-06-01:
Assuming you have the so101, navigate to `lerobot/.cache/calibration/so10`

There you will see two files: `main_leader.json` and `main_follower.json`

In those files will be a set of arrays, but the one that you'll want to check is: 
**drive_mode** which will look something like: [1,1,0,0,1,0]

_It does not matter what numbers are present,_ only that they are the same for both arms.

For example, it would be a problem if you had: 
`"drive_mode": [1,0,0,0,1,0]`
`"drive_mode": [1,1,0,0,1,0]`

...because the second motor position is a "0" in the first arm but a "1" in the second arm. **The solution is just make them both 1's or both 0's.** It doesn't seem to matter so long as they are the same.

imstevenpmwork, 2025-06-19:
Calibration process has changed significantly in the previous weeks, can you confirm you still have an issue with this?