huggingface/lerobot #1144: Depth Images

nicholscrawford, 2025-05-22:
I'm interested in using LeRobot's dataset for my own work, but it seems like depth images are not yet supported. Is that planned, or in progress? (also are there plans for documentation?)

dennisushi, 2025-05-23:
Also, segmentation images (e.g., uint16 format), and [point clouds](https://github.com/huggingface/lerobot/issues/1043). For depth images there's [a branch that appears to support them ](https://github.com/huggingface/lerobot/pull/455)

CarolinePascal, 2025-05-24:
Hi @nicholscrawford @dennisushi !

Depth images and point clouds are indeed not yet supported in LeRobot datasets. Integrating them is not planned in the near future, but we may consider it if there is enough interest from the community and enough impactful perspectives on the IA model's side.

Best,

Caroline.

Tavish9, 2025-05-25:
@nicholscrawford you can store the depth images as tensors into parquet. That's the best solution till now.

Tavish9, 2025-05-25:
> Hi [@Tavish9](https://github.com/Tavish9) !
> 
> I'm sorry we gave you this exclusivity impression :( We value all contributions from the community, but we've been dealing with huge hardware refactors lately, and didn't get time to properly address hardware related PRs.
> 
> That being said, we'll really try to make up for this once the refactors are merged !
> 
> Best,
> 
> Caroline.

I understand where lerobot’s work focus lies, but the core of many unresolved issues/prs is the lack of contributors, at least reviewers. 

For small and efficient implementations, they should be quickly accepted and everyone should be encouraged to collaboratively build the community. 
For code that affects the framework’s implementation, it can wait for the main contributor to review before making a decision.

CarolinePascal, 2025-05-25:
I completely understand and agree with your opinion. Once the refactors are merged, it will be easier to quickly merge PRs, because they will be more relevant in regard to the new codebase. Merging a PR now knowing it will be overwritten in a few days by another fix in the refactor felt a bit inefficient :/

That being said, I'm going to discuss the depth image integration with the team to see what can be done after the cameras refactor #1108 . In the meantime, **feel free to thumbs up 👍 this issue, and spread the word, to show your interest in the feature** !

CarolinePascal, 2025-11-04:
Hi @milong26, @IrvingF7,

We are currently discussing depth frames integration in [this Discord thread](https://discord.com/channels/1216765309076115607/1398650737172680810). The idea I have in mind so far is to save depth frames as uint16 grayscale PNG files during recording and to encode them using the existing pipeline as grayscale uint10 or uint12 MP4 videos.

It's still WIP for now, so feel free to come up with new ideas !

Best,

Caroline.

muhtar72603, 2025-11-14:
> > Hi [@nicholscrawford](https://github.com/nicholscrawford) [@dennisushi](https://github.com/dennisushi) !嗨，[@nicholscrawford](https://github.com/nicholscrawford)[@dennisushi](https://github.com/dennisushi) ！
> > Depth images and point clouds are indeed not yet supported in LeRobot datasets. Integrating them is not planned in the near future, but we may consider it if there is enough interest from the community and enough impactful perspectives on the IA model's side.LeRobot 数据集确实尚不支持深度图像和点云。在不久的将来没有计划整合它们，但如果社区有足够的兴趣和对 IA 模型方面有足够有影响力的观点，我们可能会考虑这样做。
> > Best,  最好
> > Caroline.  卡罗琳。
> 
> I think support for depth images or 1D images (instead of requiring 3D) would be greatly appreciated. Not only do people store depth as 1D images, but tactile readings are sometimes stored as 1D/greyscale as well.我认为对深度图像或 1D 图像（而不是需要 3D）的支持将不胜感激。人们不仅将深度存储为 1D 图像，而且触觉读数有时也存储为 1D/灰度。
> 
> Technically speaking, it's possible to simply store them as some kind of 1D tensor, but in Lerobot's many model implementations, the image keyword is scanned to decide whether a vision encoder is applied to this data, so the ability to store 1D images as `VISUAL` and `observation.images` is greatly appreciated.从技术上讲，可以简单地将它们存储为某种一维张量，但在 Lerobot 的许多模型实现中，扫描图像关键字以决定是否将视觉编码器应用于这些数据，因此将一维图像存储为 `VISUAL` 和 `observation.images` 的能力非常值得赞赏。
> 
> Best  最好

agree!!!!

atyshka, 2025-12-02:
@CarolinePascal Are you aware of this project https://github.com/cheind/hue-depth-encoding? Traditional video compression fails on depth data because it is lossy in the luminance while being pretty lossless in the chroma. This project implements a paper by the Realsense team to map depth values to color values and then encode those color images. That GitHub repo has a bunch of experiments on accuracy. I have personally used it for compressed rosbag recording and gotten over 50x compression with very little quality degradation.

atyshka, 2025-12-04:
I'll be submitting a PR this week for uncompressed UINT16 depth images. Tests are passing, I just need to confirm with real data. After that I'll explore integrating the hue-encoding compression scheme. The only thing I will note with that is that we will probably want to have control over the encoding parameters, in which case it would be good to merge #1744 first.

Also, hue encoding requires a min and max depth range (e.g. 50-5000cm) to map the colors to. We'll need to integrate this metadata with the dataset so that colorized frames can be properly decoded to depth. I'm not sure if we can store arbitrary custom metadata to the info.json but I'll give it a shot.

whats2000, 2026-04-18:
The feature is working on #3253