huggingface/lerobot #1145: Abstraction needed for user-editable configuration

brainwavecoder9, 2025-05-22:
Creating an issue for this because it has been a persistent need that merits a bigger discussion. Would like to see a solution that lets users maintain local configuration regardless of what happens with the project. 

User-related configuration data is currently buried within code, meaning that configuration changes require code changes and need to be merged/managed. The approach has also changed a few times.

It has been suggested that 777 will address this (I don't think it will). I previously made an attempt at solving this before I knew about the move to Draccus was happening (also did not address this), and Discord user @nick_builds (whose github handle is TBD) also recently made an attempt to address this. We are planning a discussion around this and would invite others to participate. 

This would ideally be a complete separation of user-managed configuration details (E.g., USB ports for arms, camera names) so that project owners could make implementation changes at-will without users having to merge or manage user-configuration data.

E.g., pulling the latest code or switching branches shouldn't change which USB port my robot is using.

brainwavecoder9, 2025-05-22:
## Wish List: 
- Single source location for all user-managed configurations (e.g., /config)
- Option for hierarchical overrides, or at least ability for users to manage local configurations at-will (different environments, different arms, etc.)
- Local files untracked, but templates are necessary (e.g., a .env.example type approach, or a folder to-be copied during setup)
- Separation to simplify user experience (e.g., the option for a small file containing only the [4 necessary key/value pairs](https://github.com/huggingface/lerobot/blob/main/lerobot/common/robot_devices/robots/configs.py) that most users need)

jccalvojackson, 2025-06-16:
reference of a setup for creating projects with the above wishlist: https://kedro.org/

jccalvojackson, 2025-06-18:
I don't like writing configurations in json or yaml since they dont provide linting (out of the box and setting it up is not trivial) so is difficult to navigate what are the options. Hence, I much rather have a user config module that is entirely written in python using dataclases (or pydantic classes). ofc, I guess it should also support for the user to change some minor flags from the cli rather than by updating the user config module.

For now, I would be happy to just have #1215 . That way, I can have my own package with my config classes which just call the relevant entrypoints of the lerobot library

jccalvojackson, 2025-09-15:
The docs have examples of writing very long cmd line options https://huggingface.co/docs/lerobot/async. Which is not ideal for the reasons above. Unfortunately, it seems that in order to support something more friendly that allows the user to write their config with python would require some significant refactor; due do dracco being used not only at the entrypoints of scripts but also in many of the class constructors.

closed 2026-03-29 by github-actions[bot]