huggingface/lerobot #1195: ros2_control support

baaluidnrey, 2025-06-03:
Hello,

I was thinking that it would be great to use the robot with ros2_control :

- to test code developped with the ROS2 framework:
- for education purposes : the robot is great, easily and not expensive to build (thank you for the work achieved), transporteable in a case, etc.

Do you have any knowledge of an existing project ?
If not, would you be interested in this kind of implementation ?

Best,
Aline

Yadunund, 2025-06-14:
@baaluidnrey having `ros2_control` integrated would be amazing!

@ycheng517 it's nice to see your PR! I'm wondering if it might be better to integrate by talking  `joint_state_broadcaster/JointStateBroadcaster`, `position_controllers/JointGroupPositionController`, `position_controllers/GripperActionController` controllers directly as opposed to integrating via `moveit2`.

We recently merged https://github.com/JafarAbdi/ros2_so_arm100/pull/15 to bringup a leader & follower SO100 arm using `ros2_controls` and I can see your PR bridging the gap to `lerobot` easily if it could skip the moveit2 layer since it simply forwards commands to the same controllers. The bringup package linked above uses this [feetech_ros2_driver](https://github.com/JafarAbdi/feetech_ros2_driver).

I'd be happy to collaborate on an effort to get this functionality merged 😄

ycheng517, 2025-06-15:
@Yadunund, I think a direct ros2_control integration would be valuable! I like MoveIt as it provides Cartesian space position and velocity control, as well as collision checking, which allows you to specify surfaces that the robot should not crashing into when executing a policy. Maybe we can incorporate both approaches. Would be happy to collaborate further!

danielsanjosepro, 2025-08-14:
@ycheng517 We also have been trying to combine ROS2 with lerobot. Nice that you also integrated moveit with lerobot!

At our lab we have implemented robot agnostic torque-based cartesian controllers (osc and cartesian impedance) for ros2_control and we built a gym environment on top of it to train VLAs and learning-based policies in general. We have been using lerobot to train policies and we have documented it in our [project website](https://utiasdsl.github.io/crisp_controllers/).

We decided to skip using MoveIt to make the interface a bit easier - just ros2_controllers that subscribe to target_pose / target_joints - and also to use the nice compliant properties of a cartesian impedance controller.
The gym however, should also work with position/velocity controlled robots, given that they also provide a topic to publish target poses or joints. We use the interface with our Franka Robotics FR3 and it has been working quite good. 

Love to hear some feedback if you try it out, cheers!

pkooij, 2026-10-21:
Thanks for the contributions here! We’re consolidating ROS 2 integration planning in #4368. Please carry over any unique requirements there. Closing this older discussion.

closed 2026-10-21 by pkooij