huggingface/lerobot #1204: example\2_calculate_validation_loss.py

Mario-kkk, 2025-06-05:
### System Info

```Shell
Error Message
IndexError: index 185 is out of bounds for dimension 0 with size 21
The error occurs because the validation dataset is created using a subset of episode indices taken from the full dataset (indices 185 to 205), but the dataset class (LeRobotDataset) likely expects local (zero-based) episode indices ranging from 0 to len(val_episodes) - 1. When it tries to use index 185 directly inside the dataset, it exceeds the size of the validation dataset, which only contains 21 episodes.
How to solve it
```

### Information

- [x] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

ep_start = self.episode_data_index["from"][ep_idx]

### Expected behavior

update code

MODKID411, 2025-06-17:
Good

zheruiqiu, 2025-06-19:
> Hey we have a big doc refactor, all those issue will be fixed, thank you for pointing it

@AdilZouitine Hi, thanks for the update! Could you please let us know when the doc refactor is expected to be released?

atyshka, 2025-06-24:
This is not a docs issue, is it? It seems to be a code issue tracked in #816