huggingface/lerobot #1237: scservo_sdk's version issue??

lanseyege, 2025-06-09:
File "lerobot_so100/common/motors/feetech/feetech.py", line 133, in __init__
    self.packet_handler = scs.PacketHandler(protocol_version)
                          ^^^^^^^^^^^^^^^^^
AttributeError: module 'scservo_sdk' has no attribute 'PacketHandler'. Did you mean: 'PortHandler'?

It seems you adopt a different version of scservo_sdk ?? There is no 'scs.PacketHandler' function, no mater scservo_sdk v1.0 or v2.0.

lanseyege, 2025-06-09:
> File "lerobot_so100/common/motors/feetech/feetech.py", line 133, in **init** self.packet_handler = scs.PacketHandler(protocol_version) ^^^^^^^^^^^^^^^^^ AttributeError: module 'scservo_sdk' has no [...]

or, did you use an inner version for scservo_sdk??

siddhu003, 2025-07-04:
How did you install scservo_sdk==1.0.0? I'm not able to find it

forwardyyy, 2025-07-21:
I met this problem too. How did you solve it?