huggingface/lerobot #1238: FileNotFoundError: [Errno 2] No such file or directory: '<my own path>/aloha_sim_dataset/aloha_sim_insertion_human_image/meta/tasks.jsonl'

xuanyaoming, 2025-06-09:
### System Info

```Shell
- `lerobot` version: 0.1.0
- Platform: Linux-5.15.0-112-generic-x86_64-with-glibc2.31
- Python version: 3.11.11
- Huggingface_hub version: 0.29.1
- Dataset version: 3.3.2
- Numpy version: 2.2.3
- PyTorch version (GPU?): 2.6.0+cu124 (True)
- Cuda version: 12040
- Using GPU in script?: <fill in>
```

### Information

- [x] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

The issue happens in `load_tasks()` function called by the constructor of `LeRobotDatasetMetadata`. Apparently the function expects a jsonl file in `meta/tasks.jsonl`, but almost all datasets on huggingface only have `meta/tasks.parquet`, which is incompatible.

How to reproduce:
1. Download `lerobot/aloha_sim_insertion_human_image` to `<my_parent_folder>/aloha_sim_dataset/aloha_sim_insertion_human_image`
2. run python code `dataset = LeRobotDataset(repo_id, root="<my_parent_folder>/aloha_sim_dataset/aloha_sim_insertion_human_image", 
                             episodes=[0, 10, 11, 23], revision="any")`

### Expected behavior

LeRobot should successfully load the local dataset in the offline mode without any error.

brainwavecoder9, 2025-06-10:
I apologize I don't recall all of the details for exactly how I solved this, and there might be a standard/better way to address this, but it is possible to generate the tasks.jsonl from tasks.parquet - something like this: 

```
import pandas as pd
import json

# Convert the specific dataset
df = pd.read_parquet("<my_parent_folder>/aloha_sim_dataset/aloha_sim_insertion_human_image/meta/tasks.parquet")
with open("<my_parent_folder>/aloha_sim_dataset/aloha_sim_insertion_human_image/meta/tasks.jsonl", 'w') as f:
    for _, row in df.iterrows():
        f.write(json.dumps(row.to_dict()) + '\n')
```

Hukongtao, 2025-07-21:
I encountered the same problem:
https://github.com/huggingface/lerobot/pull/1412#issuecomment-3095727454

Hukongtao, 2025-07-21:
```
huggingface-cli download --repo-type dataset --resume-download lerobot/aloha_mobile_cabinet --local-dir aloha_mobile_cabinet_v21 --revision v2.1
```
You can use the command above to download the v2.1 dataset

michel-aractingi, 2025-07-21:
We updated a v3.0 branch for datasets on lerobot to prepare for the dataset v3 release. 

If you download these datasets from the `main` branch it shouldn't make a difference because it will automatically pull v2.1 based on the CODEBASE_VERSION variable.

closed 2026-02-24 by github-actions[bot]