huggingface/lerobot #1239: smolvla model not working properly

yuyoujiang, 2025-06-09:
I trained an ACT with 30 episodes data, and the entire training process went smoothly without any issues. The testing process was also very smooth. However, when I switched the model to SmolVLA, the model failed to function properly during inference.

This is the dataset I used：
https://huggingface.co/datasets/youjiang97/so100_tt 

These are the commands I used for training and testing：
```bash
python lerobot/scripts/train.py   --dataset.repo_id=${HF_USER}/so100_tt   --policy.type=smolvla   --output_dir=outputs/train/smolvla_so100_0   --job_name=smolvla_so100_0   --policy.device=cuda 
```

```bash
python lerobot/scripts/control_robot.py   --robot.type=so100   --control.type=record   --control.fps=30   --control.single_task="Grasp a lego block and put it in the bin."   --control.repo_id=${HF_USER}/eval_smolvla_so100_tt   --control.tags='["tutorial"]'   --control.warmup_time_s=5   --control.episode_time_s=60   --control.reset_time_s=5   --control.num_episodes=10   --control.push_to_hub=true   --control.policy.path=outputs/train/smolvla_so100_1/checkpoints/last/pretrained_model
```

Here is the failed demo:
https://www.youtube.com/watch?v=kgVInbudhiw

Here is the training configuration file：
[train_config.json](https://github.com/user-attachments/files/20650017/train_config.json)

Any advice would be greatly appreciated.

danaaubakirova, 2025-06-10:
Hi there!

Thanks for sharing your dataset and the feedback on SmolVLA. I took a look, it's quite complex setup, and with only 30 episodes and varying object positions, it's tough for current robotics models to learn robustly.

I recommend checking out this dataset for reference:
🔗 [SVLA SO100 PickPlace](https://huggingface.co/spaces/lerobot/visualize_dataset?path=%2Flerobot%2Fsvla_so100_pickplace%2Fepisode_0)

In this dataset, we recorded 50 episodes across 5 distinct cube positions. For each position, we collected 10 episodes of pick-and-place interactions. This structure, repeating each variation several times, helped the model generalize better. We tried similar dataset with 25 episodes, and it was not enough leading to a bad performance. So, the data quality and quantity is definitely a key!

kasiv008, 2025-06-15:
A small update. As suggested by @danaaubakirova, I increased the number of episodes to 80 and got SmolVLA to working!
Thanks.

chirag1701, 2025-07-05:
> But I record 100 datasets, i am sad

I faced the same problem with 50 Episodes on pick_place (https://huggingface.co/chirag1701/so101_pickplace_B_smolvla_cwnu)
I will now try with 90 episodes. Any update from your side? Did you solve the problem?

ZhuYaoHui1998, 2025-07-11:
no, its still not work.

chirag1701, 2025-07-12:
I have had success with smolvla with making many episodes for the same position. And make for multiple positions. Don't make just 1 epsiode for 1 position

Liu-Xiaoyan97, 2025-07-23:
> no, its still not work.

did you solve the problem?

chirag1701, 2025-07-23:
@Liu-Xiaoyan97 In your case, i think the problem is with the camera positions you have taken. There should be a full context view for the robot to see the whole broad setup. Thats why its suggested to keep one "up" camera for the full view and the other "gripper" for precision in picking the blocks. But, both of the cameras in your case almost give similar views and hence the model is not able to pinpoint the location of the blocks. Let me know if you face any problem :)

chirag1701, 2025-07-23:
[chirag1701/new0_green_final](https://huggingface.co/datasets/chirag1701/new0_green_final 
refer to this dataset for reference

danaaubakirova, 2025-08-05:
Thanks everyone for sharing your datasets and observations. It's been really helpful to understand the different cases where SmolVLA struggles.

**TL;DR:**
SmolVLA is more sensitive than ACT to data quality, episode length, and camera setup. If actions are too fast, or visual coverage is limited (e.g., no top view in [Xiaoyan97/orange_block_pickplace](https://huggingface.co/datasets/Xiaoyan97/orange_block_pickplace)), the model tends to fail during inference, e.g. getting stuck in loops or showing shaky behavior. This usually indicates failure to properly learn during training.

**Why short actions or fast episodes can cause failure?**
From our experience, very short episodes with quick actions often result in poor learning.
When the robot moves too quickly, the model receives fewer frames to understand how actions affect the state. This weakens the ability to learn meaningful transitions. Attaching some visuals examples from what you have shared here: 

Good example:
[Episode 99](https://huggingface.co/spaces/lerobot/visualize_dataset?path=%2Fchirag1701%2Fnew0_green_final%2Fepisode_99) – ~16 seconds (~480 frames at 30 FPS)
→ Smooth, well-paced motion with clear progression. Easier for the model to learn.

Less effective example:
[Episode 0](https://huggingface.co/spaces/lerobot/visualize_dataset?path=%2FXiaoyan97%2Forange_block_pickplace%2Fepisode_0) – ~10 seconds (~300 frames)
→ Too short and fast, with abrupt joint changes. Harder for the model to extract motion cues.

**Recommendations for better SmolVLA performance**

1. Record actions slowly and smoothly. Avoid fast or jerky motions.

2. Don’t collect just one episode per object position. Use 5–10 episodes per position across unique positions you want to have.

3. Use a top-view camera that captures the full workspace, so the model sees the full spatial context (arm, object layout, goal area). Use a gripper-view camera for precision. If both cameras are too similar (e.g., both zoomed-in), the model may fail to localize objects reliably.

I know this may sound like a lot of setup rules, but it's all part of building more robust VLA models. The better our datasets, the better these models will generalize and perform :)

If your dataset is already recorded:

- Consider filtering out very short or abrupt episodes
- Make sure action and observation signals are properly aligned
- Watch for issues like delayed or inconsistent gripper signals

More data recording tips are available here:
🔗 https://huggingface.co/blog/lerobot-datasets

Feel free to share specific episodes or demos if you'd like feedback or help debugging.