huggingface/lerobot #1242: SmolVLA Gym Simulation - Release?

Jaykumaran, 2025-06-09:
Hello,

I've trained the smolvla_base for 200K steps. I'm trying to do a inference and visualize like we do for aloha or pusht. Could anyone guide me on this. 

I dont have a robot arm, so Gym simulation is something I'm looking for, when will it be released?

zttiannnn, 2025-06-10:
I tried to use the lerobot/aloha_sim_insertion_human dataset to train smolvla_base and simulate it in the aloha Gym simulation. However, the inference effect is poor. I'm not sure if this attempt is feasible. Any reply would be helpful!

zttiannnn, 2025-06-10:
To be honest, I'm also new to this, but I don't think the dataset configs should be a crucial issue. 🤔 Thanks for your sharing. I'll go and consult him.

JuilieZ, 2025-06-11:
> I tried to use the lerobot/aloha_sim_insertion_human dataset to train smolvla_base and simulate it in the aloha Gym simulation. However, the inference effect is poor. I'm not sure if this attempt is feasible. Any reply would be helpful!

Hi,have you ever found the simulation inference for SmolVLA? 
Actually I only find the DiffusionPolicy inference o(╥﹏╥)o

JuilieZ, 2025-06-11:
> I think you're right. I checked the pretraining dataset, and most of the robots are so100. I didn't even see aloha. So it may not be possible to use the simulation environment to infer and view the model's performance.

actually I even dont know how to inference so100 in sim env, there is no smolVLA inference for sim env right?

Jaykumaran, 2025-06-11:
@zttiannnn

> I tried to use the lerobot/aloha_sim_insertion_human dataset to train smolvla_base and simulate it in the aloha Gym simulation. However, the inference effect is poor. I'm not sure if this attempt is feasible. Any reply would be helpful!

Ca you share the script that you have tried.

Also, the modelling part of smolvla quite a time uses adapt_to_pi_aloha

https://github.com/huggingface/lerobot/blob/main/lerobot/common/policies/smolvla/modeling_smolvla.py

zttiannnn, 2025-06-12:
I just tried train.py and eval.py in lerobot/scripts. For example, python lerobot/scripts/train.py   --policy.path=lerobot/smolvla_base   --dataset.repo_id=lerobot/aloha_sim_insertion_human   --batch_size=64   --steps=20000. When running eval.py, select the aloha environment: --env.type=aloha

JuilieZ, 2025-06-13:
> I just tried train.py and eval.py in lerobot/scripts. For example, python lerobot/scripts/train.py --policy.path=lerobot/smolvla_base --dataset.repo_id=lerobot/aloha_sim_insertion_human --batch_size=64 --steps=20000. When running eval.py, select the aloha environment: --env.type=aloha
have you erver tried shorter chunk size? like 20?

Jaykumaran, 2025-06-16:
Hey,

Do watch LeRobot hackathon solutions by various teams, they are supercool. 

1. https://x.com/suessmannn/status/1934361919755813208

P.S> This msg is not relevant to this thread's topic.

chenkang455, 2025-07-24:
I am tring pi0 on the aloha, it has no performance too...

https://github.com/user-attachments/assets/f37bbd7a-6a1a-41ef-ae50-52e66c604dff