huggingface/lerobot #1245: Press Enter does not work on Windows in calibration process

B-Rousseau, 2025-06-09:
Calibration script is not multi-platforms, it does not work on Windows due to the code of "Press Enter".

Please consider this modifications in _**\lerobot\common\utils\utils.py**_ :

**Add** : import keyboard

**Replace** :
def enter_pressed() -> bool:
    return select.select([sys.stdin], [], [], 0)[0] and sys.stdin.readline().strip() == ""

**By** :
def enter_pressed() -> bool:
    return keyboard.is_pressed("enter")

B-Rousseau, 2025-06-14:
> I also have the same question，Have you solved it yet？

Yes I solved it with thé code in my ticket.
Cordialement,
Benoit Rousseau.

BelieveScience, 2025-06-14:
这是来自QQ邮箱的假期自动回复邮件。你好，我最近正在休假中，无法亲自回复你的邮件。我将在假期结束后，尽快给你回复。