huggingface/lerobot #1252: Failed to sync read 'Present_Position' on ids=[2,3,4,6]after 1 tries. [TxRxResult] There is no status packet

huazai665, 2025-06-10:
my arm is koch，when I set the motors ids and baudrates, it report error:
Failed to sync read 'Present_Position' on ids=[2,3,4,6]after 1 tries. [TxRxResult] There is no status packet

banisadr, 2025-06-15:
Facing the same error, teleop is working fine but only record is failing

huazai665, 2025-06-16:
> [@huazai665](https://github.com/huazai665) [@banisadr](https://github.com/banisadr)
> 
> I tried the following solution and it worked.
> 
> lerobot/common/robots/koch_follower/koch_follower.py, line 179 obs_dict = self.bus.sync_read("Present_Position") -> obs_dict = self.bus.sync_read("Present_Position", num_retry=3)
> 
> I started adjusting the num_try parameter from 1, and it worked when it was 3. It might be safer if it's bigger. But I'm not sure what effect increasing this parameter will have. I guess it may reduce the overall efficiency.

Thanks, we modified "obs_dict = self.bus.sync_read("Present_Position") -> obs_dict = self.bus.sync_read("Present_Position", num_retry=6)", and it seems to work. But there is a problem for us. The middle motor seems wrong, for example, when I make the master arm move upper, the follower arm moves down.

skalade, 2025-06-18:
> Thanks, we modified "obs_dict = self.bus.sync_read("Present_Position") -> obs_dict = self.bus.sync_read("Present_Position", num_retry=6)", and it seems to work. But there is a problem for us. The [...]

I ran into the same issue with the koch follower and the solution was to enable drive_mode for dynamixel bus... In https://github.com/huggingface/lerobot/blob/main/lerobot/common/motors/dynamixel/dynamixel.py#L110 change to `apply_drive_mode = True`.

huazai665, 2025-06-20:
> I think the only time I've seen that error was when I didn't put the arm in it's "middle" range of motion position during calibration. https://huggingface.co/docs/lerobot/so101#calibration-video <- are you calibrating it like in the video?
> 
> Not sure what else it could be without more info.

@skalade, yes, I calibrate it according to the video. When I put the arm in its middle， it reports 'ValueError: Value -4294964294 out of range for 4-byte two's complement : [-2147483648, 2147483647]. Before I calibrate the arms, I have made an ID for each motor. But if I change the "apply_drive_mode=False", it does not report this error.

ultrafro, 2025-09-01:
i faced a similar issue with a WOWROBO kit while using a 5v supply. The WOWROBO kit uses 12v servos. The problems resolved for me after switching to a 12V supply

CursedRock17, 2026-02-03:
### Environment
**OS**: Ubuntu 24.04
**Python**: 3.10
**Lerobot Version**: 0.4.2
**Platform**: SO101 Arms

### Issue + Resolution
I've run into this issue on many occasions specifically in the `lerobot-record` function, in both the Python/Shell variants. When teleoperating, I could do crazy movements and this error wouldn't throw, same with the [Qt_Debugger](https://github.com/Kotakku/FT_SCServo_Debug_Qt) that Feetech provides for their [ST3215](https://www.seeedstudio.com/Feetech-ST-3215-C044-Heavy-Duty-Servo-7-4V-1-191-Gear-Reduction-p-6460.html) servos.

I had 2 Cameras and 2 SO101 Arms (Leader/Follower) all plugged into 1 Cheap USB hub and it severely amplified the issue, especially when the follower arm wasn't home at the end of the episode. On my machine, I dedicated 1 USB port to each of the cameras, then 1 to the hub for both of the arms (3 total) and the issue showed up less. Unfortunately it was still occuring every ~15 episodes, so I had to up the default amount of `num_retry` attempts to 5 as a sort of patch, [here](https://github.com/huggingface/lerobot/blob/v0.4.2/src/lerobot/motors/motors_bus.py#L1053) and [here](https://github.com/huggingface/lerobot/blob/v0.4.2/src/lerobot/motors/motors_bus.py#L1101). Hopefully this serves useful to somebody.