huggingface/lerobot #1270: smolvla not work！

ZhuYaoHui1998, 2025-06-12:
### System Info

```Shell
Copy-and-paste the text below in your GitHub issue and FILL OUT the last point.

- `lerobot` version: 0.1.0
- Platform: Linux-6.8.0-60-generic-x86_64-with-glibc2.35
- Python version: 3.10.13
- Huggingface_hub version: 0.30.2
- Dataset version: 3.5.0
- Numpy version: 1.26.0
- PyTorch version (GPU?): 2.6.0+cu124 (True)
- Cuda version: 12040
- Using GPU in script?: <fill in>
```

### Information

- [ ] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

I collected 100 sets of data where a green block was placed into a fixed basket, with the green block randomly positioned. Two cameras were used. However, SmoLVLA still keeps shaking and looping head movements at the initial position after 20k training steps, even though the loss has dropped to 0.021. How can I fix this? Under the same dataset, ACT works fine.

### Expected behavior

I collected 100 sets of data where a green block was placed into a fixed basket, with the green block randomly positioned. Two cameras were used. However, SmoLVLA still keeps shaking and looping head movements at the initial position after 20k training steps, even though the loss has dropped to 0.021. How can I fix this? Under the same dataset, ACT works fine.

ZhuYaoHui1998, 2025-06-12:
https://huggingface.co/datasets/Marlboro1998/starai02  this is my datasets

ChenLu-china, 2025-06-13:
Hello, I also tried SmolVLA for the double so100, the loss only dropped to 0.015 and kept it on.

n3puiol, 2025-06-13:
I am unsure as the effect it could have but you could try messing with the normalization mapping. I've noticed some interesting changes in xarm environment if I set action to `min_max`.

```py
normalization_mapping = {
    FeatureType.VISUAL: NormalizationMode.IDENTITY,
    FeatureType.STATE: NormalizationMode.MEAN_STD,
    FeatureType.ACTION: NormalizationMode.MIN_MAX
}
```

niteshsilwal01, 2025-06-27:
@ZhuYaoHui1998 could you please share what steps you followed for testing the trained smolVLA policy, and how did you apply it to the real robot?