huggingface/lerobot #1271: Cannot run evaluation with real robot. (with `lerobot/record.py`)

kaaatsu32329, 2025-06-12:
I would like to run `lerobot/record.py` and evaluate the policy on the real robot (so101) based on the pretrained policy, but I get the following message.

Run below,

```bash
python -m lerobot.record \
    --robot.type=so101_follower \
    --robot.port=/dev/ttyACM1 \
    --robot.calibration_dir=.cache \
    --robot.id=armer_follower \
    --robot.cameras="{wrist: {type: opencv, index_or_path: 0, width: 640, height: 480, fps: 30}}" \
    --policy.path=outputs/train/${policy_name}/${file_name}/checkpoints/last/pretrained_model \
    --policy.device=cuda \
    --dataset.fps=30 \
    --dataset.root=${root} \
    --dataset.repo_id="so_101/log" \
    --dataset.tags='["so101"]' \
    --dataset.episode_time_s=30 \
    --dataset.reset_time_s=40 \
    --dataset.num_episodes=100 \
    --dataset.push_to_hub=false \
    --dataset.single_task="Pick and place"
```

and the following

```bash
Traceback (most recent call last):
  File "/home/user/miniconda3/envs/soarm101/lib/python3.10/site-packages/draccus/parsers/decoding.py", line 159, in decode_dataclass
    instance = origin(**init_args)
  File "<string>", line 10, in __init__
  File "/home/user/arm_ws/soarm101/lerobot/lerobot/record.py", line 143, in __post_init__
    raise ValueError("Choose either a policy or a teleoperator to control the robot")
ValueError: Choose either a policy or a teleoperator to control the robot
```

Upon verification, it seemed to be False==False here.
https://github.com/huggingface/lerobot/blob/edfebd522c34b9ad4763fb078304b09dce360b04/lerobot/record.py#L141-L144

siddhu003, 2025-06-17:
Can you tell me which policy you're using and if it works as expected?

Avory1258, 2025-06-18:
I encountered the same problem as you, how did you solve it?

siddhu003, 2025-06-19:
Yes, I tried increasing the number of the steps. I tried it after training for about 150,000 steps. I made little to no difference. I tried obstructing both the cameras feed by placing my hands infront of them and it behaved exactly the same. That's the reason I think there might be issues with configuring the cameras. I mean, is there somewhere in the scripts where I should change how the camera feed is read?

Did you finetune your model on your private dataset or some publicly available dataset?

kaaatsu32329, 2025-06-19:
There is a `dataset.video_backend` option regarding the video. (I don't know if they are related.)
And I use private datasets.

Avory1258, 2025-06-20:
My robot basically cannot grab the block, it always stays one body length away from it. Also, its running time always stops at 50 seconds, I am not sure if it is because the data set I recorded is 50 seconds, can you maybe help me solve this problem?

kaaatsu32329, 2025-06-26:
You can change running time with arguments.

```
python -m lerobot.record \
    ...
    --dataset.episode_time_s=100 \
    --dataset.reset_time_s=100
```