huggingface/lerobot #1329: Size mismatch error when evaluating a fine-tuned VLA model

RonanFR, 2025-06-16:
### System Info

```Shell
- `lerobot` version: 0.1.0
- Platform: Linux-6.8.0-60-generic-x86_64-with-glibc2.39
- Python version: 3.10.16
- Huggingface_hub version: 0.30.2
- Dataset version: 3.5.0
- Numpy version: 2.2.4
- PyTorch version (GPU?): 2.6.0+cu124 (True)
- Cuda version: 12040
- Using GPU in script?: yes
```

### Information

- [x] One of the scripts in the examples/ folder of LeRobot
- [x] My own task or dataset (give details below)

### Reproduction

I have bought all the parts from Seeed Studio for building SO arm 100.
I have successfully assembled the 2 arms (leader + followers), calibrated them, etc... following this documentation: https://wiki.seeedstudio.com/lerobot_so100m/

Following the exact same documentation, I have:
- recorded a dataset (pushed to the HF Hub: https://huggingface.co/datasets/rfruit/so100_building_house
- fine-tuned 2 models: https://huggingface.co/rfruit/lerobot_imitation_building_house (SmolVLA) and https://huggingface.co/rfruit/act_building_house (ACT)
- tried to evaluate the fined-tuned model but the following error is triggered (for both models):
```RuntimeError: Error(s) in loading state_dict for SmolVLAPolicy:
    size mismatch for normalize_inputs.buffer_observation_state.mean: copying a param with shape torch.Size([32]) from checkpoint, the shape in current model is torch.Size([6]).
    size mismatch for normalize_inputs.buffer_observation_state.std: copying a param with shape torch.Size([32]) from checkpoint, the shape in current model is torch.Size([6]).
    size mismatch for normalize_targets.buffer_action.mean: copying a param with shape torch.Size([32]) from checkpoint, the shape in current model is torch.Size([6]).
    size mismatch for normalize_targets.buffer_action.std: copying a param with shape torch.Size([32]) from checkpoint, the shape in current model is torch.Size([6]).
    size mismatch for unnormalize_outputs.buffer_action.mean: copying a param with shape torch.Size([32]) from checkpoint, the shape in current model is torch.Size([6]).
    size mismatch for unnormalize_outputs.buffer_action.std: copying a param with shape torch.Size([32]) from checkpoint, the shape in current model is torch.Size([6])```

I models were fine-tuned using a Hugging Face Space but the same error occurs when I fine-tune on my own computer (the same computer that I used to record the dataset and that I am using to try to evaluate the model).

The Seeed Studio documentation suggests to clone the fork https://github.com/ZhuYaoHui1998/lerobot but I also tried to clone and install the original lerobot repo and the same error is triggered.

### Expected behavior

I would expect that the size of all tensors of the fine-tuned model (ACT or SmolVLA) match the size of the model that is loaded for evaluation.

Sergeytata, 2025-06-16:
If you used merge.py script. You need to select the correct max_depth. For a single arm it’s 6, by default it’s 32. You need to rerun the script with correction. 
A possible easy fix. Load the weights using safetensor, slice them manually and save. Load the model from the weights and train with corrected dataset for a few steps (couple thousands) without warmup and with low lr. This should recover your model.

sungjuune, 2025-06-22:
Thank you @Sergeytata !!
Can you please give us more details for your solution? I also faced the same error when evaluating the fine-tuned SmolVLA model, but I didn't use `merge.py`. I just ran the script:
```
python lerobot/scripts/train.py \
  --policy.path=lerobot/smolvla_base \
  --dataset.repo_id=lerobot/svla_so100_stacking \
  --batch_size=64 \
  --steps=20000
```
for fine-tuning. Then when I run evaluation:
```
python lerobot/scripts/eval.py \
--policy.path=outputs/train/XXX/XXX_smolvla/checkpoints/last/pretrained_model \
 --env.type=aloha
```
I get the following error:
```
RuntimeError: Error(s) in loading state_dict for SmolVLAPolicy:
        size mismatch for normalize_inputs.buffer_observation_state.mean: copying a param with shape torch.Size([6]) from checkpoint, the shape in current model is torch.Size([14]).
        size mismatch for normalize_inputs.buffer_observation_state.std: copying a param with shape torch.Size([6]) from checkpoint, the shape in current model is torch.Size([14]).
        size mismatch for normalize_targets.buffer_action.mean: copying a param with shape torch.Size([6]) from checkpoint, the shape in current model is torch.Size([14]).
        size mismatch for normalize_targets.buffer_action.std: copying a param with shape torch.Size([6]) from checkpoint, the shape in current model is torch.Size([14]).
        size mismatch for unnormalize_outputs.buffer_action.mean: copying a param with shape torch.Size([6]) from checkpoint, the shape in current model is torch.Size([14]).
        size mismatch for unnormalize_outputs.buffer_action.std: copying a param with shape torch.Size([6]) from checkpoint, the shape in current model is torch.Size([14]).
```

Thank you!