huggingface/lerobot #1346: LeRobotDataset May Eagerly Load Images, Making It Unusable for Large Datasets

klingmad, 2025-06-18:
### System Info

```Shell
Hello LeRobot Team! Thank's for providing this great repository and the dataset visualizer.

Insofar as I understand it, LeRobotDataset always keeps the whole dataset in memory, which causes out-of-memory issues and makes it unusable with datasets that exceed the machine's memory. Are there any plans to provide tooling to "add" episodes to existing datasets, without keeping the whole dataset in memory and for lazy loading? The PyTorch DataLoader, for example, also require the LeRobotDatset to be in memory.

Out-of-memory issues occur quite quickly when using datasets required to fine-tune PI0 and GR00T. For example, an image with 256x256 pixels and 3 channels, where each pixel is 1 byte, requires 256*256*3 = 196,608 bytes = 192KB. GR00T 1.5 suggests a fine-tuning configuration with a large batch size, say 128, and 20K steps. If each frame has two images, the dataset has 5.12 million images. This requires 937.5 GB which exceeds DGX H100 memory (8 * H100).

The call stack is:
`save_episode` saves the current episode and calls `save_episode_table` which calls both:
1. `embed_images` embeds image bytes into the table
2. ep_dataset = embed_images(ep_dataset)
   self.hf_dataset = concatenate_datasets([self.hf_dataset, ep_dataset])
and `hf_dataset` stays in memory.
```

### Information

- [ ] One of the scripts in the examples/ folder of LeRobot
- [x] My own task or dataset (give details below)

### Reproduction

ep_dataset = embed_images(ep_dataset)
  self.hf_dataset = concatenate_datasets([self.hf_dataset, ep_dataset])
and `hf_dataset` stays in memory.

### Expected behavior

After saving the episode to disk, the self.hf_dataset should be cleared or save only references to the parquet files, instead of accumulating the pngs.

Tavish9, 2025-06-19:
try to save the dataset in "video" mode, so that it would load the images just when it needs to.

dangr, 2025-08-26:
For saving episodes to disk, simply removing the `concatenate_datasets` line worked for me.  The resulting files are identical, and no OOM error.

A temporary workaround might be to add a 'don't accumulate hf_dataset' flag

davidtao212, 2025-10-31:
> For saving episodes to disk, simply removing the `concatenate_datasets` line worked for me. The resulting files are identical, and no OOM error.
> 
> A temporary workaround might be to add a 'don't accumulate hf_dataset' flag

@dangr Cool, thanks for your suggestion. I just monkey pathed as follows and compared the SHA of generated ds with that of unpatched ones, it works perfectly well.
```python
import datasets
from lerobot.common.datasets.utils import embed_images, hf_transform_to_torch

def _save_episode_table_patch(self, episode_buffer: dict, episode_index: int) -> None:
    episode_dict = {key: episode_buffer[key] for key in self.hf_features}
    ep_dataset = datasets.Dataset.from_dict(episode_dict, features=self.hf_features, split="train")
    ep_dataset = embed_images(ep_dataset)
    self.hf_dataset.set_transform(hf_transform_to_torch)
    ep_data_path = self.root / self.meta.get_data_file_path(ep_index=episode_index)
    ep_data_path.parent.mkdir(parents=True, exist_ok=True)
    ep_dataset.to_parquet(ep_data_path)

LeRobotDataset._save_episode_table = _save_episode_table_patch
```

klingmad, 2025-10-31:
The new LeRobot versions don't struggle with this so I'll close this issue

closed 2025-10-31 by klingmad