huggingface/lerobot #1351: Need help about dataset and train.

hbj52152, 2025-06-19:
# What this for

Attracted by smolvla, and new to smolvla_base, and i am now trying to ask few questions before a try with this model.

Several parts: 
1) dataset
2) simulation
3) real world

## dataset
### Two cameras ?
I have read three datasets, including  
https://huggingface.co/datasets/lerobot/svla_so101_pickplace
https://huggingface.co/datasets/Marlboro1998/starai02

and its structure shows: 
videos/chunks/ two foldes with .mp4 files, each is one camera.

https://huggingface.co/datasets/unitreerobotics/Z1_DualArmStackBox_Dataset

I find that the data in unitree dataset is with one camera

does it mean that it is not necessary with two cameras? 

**if one camera** is possible to build dataset. Where and how should i change the code to build the dataset and to train with it? 

**if two cameras are min demand**, is it possible i make it with random position? like one is in-hand, and one is some where else, because it might be hard to real put it everytime in the same position ( for some work)

### depth data?
I have one realsense camera with depth data. how should i deal with it in dataset? only with color frame?

### video length
I have watch several videos in svla_so101_pickplace, and each is with 10s. I understand that this is because such shot video contains a complete task. 

how about a work might be long and complex? break it down into n parts so you will get n +1 (break down + full) tasks and then train with it?

## simulation

### simulation env
i got some basic understanding in this part. I used few times with mujoco and isaac sim. just start to try with lerobot. 

Is it possible output to mujoco or isaac sim? I understand these are two might not relate to lerobot, sorry if anything wrong.

### simulation of different robot
This is something relating to train. How can i record a dataset for custom robot? I have read some dataset like for unitree, but like how to record in simulate and with custom robot?

I have not yet deep read the documentation with lerobot, so if there is any doc can help this, could you share some information.

# real world

if i try to train with other robot, but with few dataset ( because less community data and self-collection data) , i think its performance would not be as good as those in your paper. so how many data do you think necessary for such situation ( robot different from paper)

Thanks a lot for your consideration. Forgive me if anything wrong in my text above.

gujiacheng, 2025-06-20:
mark

hbj52152, 2025-06-30:
> Hi [@hbj52152](https://github.com/hbj52152) [@gujiacheng](https://github.com/gujiacheng), you can find most of the information in [any4lerobot](https://github.com/Tavish9/any4lerobot), including datasets, different robots, etc..

Thanks a lot, I will turn to it right now.