huggingface/lerobot #1368: OpenCV camera connection fails on Windows due to default backend

myanvoos, 2025-06-23:
### System Info

```Shell
- `lerobot` version: 0.1.0
- Platform: Windows-10-10.0.22631-SP0
- Python version: 3.10.18
- Huggingface_hub version: 0.33.0
- Dataset version: 3.6.0
- Numpy version: 2.2.6
- PyTorch version (GPU?): 2.7.1+cpu (False)
- Cuda version: N/A
- Using GPU in script?: No
```

### Information

- [ ] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

1. Install `lerobot` and its dependencies on a Windows machine with at least one connected webcam.
2. Ensure the full `opencv-python` package is installed.
3. Run `python lerobot/find_cameras.py opencv`.

The script should print the list of detected cameras, followed by connection errors:
```
--- Detected Cameras ---
Camera #0:
Name: OpenCV Camera @ 0
Type: OpenCV
Id: 0
Backend api: MSMF
...
Camera #1:
Name: OpenCV Camera @ 1
Type: OpenCV
Id: 1
Backend api: MSMF
...
ERROR:main:Failed to connect or configure OpenCV camera 0: Failed to open OpenCVCamera(0).Run python -m lerobot.find_cameras opencv to find available cameras.
ERROR:main:Failed to connect or configure OpenCV camera 1: Failed to open OpenCVCamera(1).Run python -m lerobot.find_cameras opencv to find available cameras.
WARNING:main:No cameras could be connected. Aborting image save.
```

I managed to resolve this issue locally by making a one-line change to `lerobot/common/cameras/opencv/camera_opencv.py`.

In the `connect` method, changing:
`self.videocapture = cv2.VideoCapture(self.index_or_path, self.backend)`

to the hardcoded:
`self.videocapture = cv2.VideoCapture(self.index_or_path, cv2.CAP_MSMF)`

forces OpenCV to use the Media Foundation backend, which then works.

### Expected behavior

It should correctly give the list of cameras and successfully capture images. For example, after hardcoding in `cv2.CAP_MSMF`:

```
--- Detected Cameras ---
Camera #0:
  Name: OpenCV Camera @ 0
  Type: OpenCV
  Id: 0
  Backend api: MSMF
  Default stream profile:
    Format: -1.0
    Width: 640
    Height: 480
    Fps: 30.0
--------------------
Camera #1:
  Name: OpenCV Camera @ 1
  Type: OpenCV
  Id: 1
  Backend api: MSMF
  Default stream profile:
    Format: -1.0
    Width: 640
    Height: 480
    Fps: 30.0
--------------------

Finalizing image saving...
Image capture finished. Images saved to outputs\captured_images
```

ashutoshsingh0223, 2025-06-26:
For me `cv2.CAP_DSHOW` is the one that worked. Not `cv2.CAP_MSMF`.

communitywireless, 2025-11-17:
Using the latest and greatest version I'm aware of - 0.4.2 - this is still a problem.

When I run other python scripts to test the connection to the cameras I can connect to the cameras and pull data.  Same from the windows camera app.

Unfortunately, the newest Windows installation or lerobot packages/compiles these scripts into executables and so, as near as I can tell, they can't be edited on the fly.  Or at least they can't be edited on the fly with my current level of knowledge about how lerobot for Windows is compiled and managed.

The following script works fine.  Connects to the cameras.  All is good.

```
import cv2
import time

for i in range(4):
    print(f"Testing camera index {i}...")
    cap = cv2.VideoCapture(i)
    if cap.isOpened():
        ret, frame = cap.read()
        if ret:
            print(f"  ✅ Camera {i}: SUCCESS (Grabbed a frame of shape {frame.shape})")
        else:
            print(f"  ❌ Camera {i}: Failed to grab a frame.")
        cap.release()
    else:
        print(f"  ❌ Camera {i}: Failed to open.")
    print("-" * 20)
    time.sleep(1) # Give camera time to release
```

Here is the configuration of my system.

```
- lerobot version: 0.4.2
- Platform: Windows-10-10.0.26100-SP0
- Python version: 3.10.19
- Huggingface Hub version: 0.35.3
- Datasets version: 4.1.1
- Numpy version: 2.2.6
- PyTorch version: 2.7.1+cpu
- Is PyTorch built with CUDA support?: False
- Cuda version: N/A
- GPU model: N/A
- Using GPU in script?: No

```

And here is how it fails:

```
>>lerobot-find-cameras opencv
[ERROR:0@2.621] global obsensor_uvc_stream_channel.cpp:163 cv::obsensor::getStreamChannelGroup Camera index out of range
[ERROR:0@2.625] global obsensor_uvc_stream_channel.cpp:163 cv::obsensor::getStreamChannelGroup Camera index out of range
[ERROR:0@2.627] global obsensor_uvc_stream_channel.cpp:163 cv::obsensor::getStreamChannelGroup Camera index out of range
[ERROR:0@2.629] global obsensor_uvc_stream_channel.cpp:163 cv::obsensor::getStreamChannelGroup Camera index out of range
[ERROR:0@2.631] global obsensor_uvc_stream_channel.cpp:163 cv::obsensor::getStreamChannelGroup Camera index out of range
[ERROR:0@2.634] global obsensor_uvc_stream_channel.cpp:163 cv::obsensor::getStreamChannelGroup Camera index out of range
[ERROR:0@2.636] global obsensor_uvc_stream_channel.cpp:163 cv::obsensor::getStreamChannelGroup Camera index out of range
[ERROR:0@2.639] global obsensor_uvc_stream_channel.cpp:163 cv::obsensor::getStreamChannelGroup Camera index out of range
[ERROR:0@2.641] global obsensor_uvc_stream_channel.cpp:163 cv::obsensor::getStreamChannelGroup Camera index out of range
[ERROR:0@2.647] global obsensor_uvc_stream_channel.cpp:163 cv::obsensor::getStreamChannelGroup Camera index out of range
[ERROR:0@2.649] global obsensor_uvc_stream_channel.cpp:163 cv::obsensor::getStreamChannelGroup Camera index out of range
[ERROR:0@2.652] global obsensor_uvc_stream_channel.cpp:163 cv::obsensor::getStreamChannelGroup Camera index out of range
[ERROR:0@2.655] global obsensor_uvc_stream_channel.cpp:163 cv::obsensor::getStreamChannelGroup Camera index out of range
[ERROR:0@2.659] global obsensor_uvc_stream_channel.cpp:163 cv::obsensor::getStreamChannelGroup Camera index out of range
[ERROR:0@2.663] global obsensor_uvc_stream_channel.cpp:163 cv::obsensor::getStreamChannelGroup Camera index out of range
[ERROR:0@2.665] global obsensor_uvc_stream_channel.cpp:163 cv::obsensor::getStreamChannelGroup Camera index out of range
[ERROR:0@2.667] global obsensor_uvc_stream_channel.cpp:163 cv::obsensor::getStreamChannelGroup Camera index out of range
[ERROR:0@2.672] global obsensor_uvc_stream_channel.cpp:163 cv::obsensor::getStreamChannelGroup Camera index out of range
[ERROR:0@2.675] global obsensor_uvc_stream_channel.cpp:163 cv::obsensor::getStreamChannelGroup Camera index out of range
[ERROR:0@2.679] global obsensor_uvc_stream_channel.cpp:163 cv::obsensor::getStreamChannelGroup Camera index out of range
[ERROR:0@2.681] global obsensor_uvc_stream_channel.cpp:163 cv::obsensor::getStreamChannelGroup Camera index out of range
[... 70 lines cut ...]
    Height: 480
    Fps: 0.0
--------------------
Camera #3:
  Name: OpenCV Camera @ 3
  Type: OpenCV
  Id: 3
  Backend api: DSHOW
  Default stream profile:
    Format: -1.0
    Fourcc: YUY2
    Width: 640
    Height: 480
    Fps: 0.0
--------------------
ERROR:lerobot.scripts.lerobot_find_cameras:Failed to connect or configure OpenCV camera 0: Failed to open OpenCVCamera(0).Run `lerobot-find-cameras opencv` to find available cameras.
ERROR:lerobot.scripts.lerobot_find_cameras:Failed to connect or configure OpenCV camera 1: Failed to open OpenCVCamera(1).Run `lerobot-find-cameras opencv` to find available cameras.
ERROR:lerobot.scripts.lerobot_find_cameras:Failed to connect or configure OpenCV camera 2: Failed to open OpenCVCamera(2).Run `lerobot-find-cameras opencv` to find available cameras.
ERROR:lerobot.scripts.lerobot_find_cameras:Failed to connect or configure OpenCV camera 3: Failed to open OpenCVCamera(3).Run `lerobot-find-cameras opencv` to find available cameras.
WARNING:lerobot.scripts.lerobot_find_cameras:No cameras could be connected. Aborting image save.
```

Since I can't just hack in the work-around using this installation method, I think I'm going to have to remove the software and start over.

I've been hacking away at this for a couple of days and I can't get through one iteration of the workflow.   Between versioning errors, incompatible models, re-hacking the Colab training example ( which doesn't work at all without significant work ) incorrect commands on the default card that the training pushes to Hugging Face, this is starting to get really frustrating.