huggingface/lerobot #1370: SmolVLA have poor performance

FanZhang91, 2025-06-23:
### System Info

```Shell
python scripts/train.py --policy.type=smolvla --dataset.repo_id=self_data --batch_size=64 --steps=100000
```

### Information

- [ ] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

I use 180 episode data(mobile robot dataset, forklift dataset) to train a model. Using the same data, the pi0 model fit the training data well。But the SmolVLA's prediction results are quite different from GT action.
file:///home/zhangfan/deep_learning/SmolVLA/lerobot/serving_infer/put_next_action.jpg

I change both state and action normalization method to NormalizationMode.MIN_MAX in training phase. 

How to fix this problem?

### Expected behavior

None

hahans, 2025-06-23:
I think part of the reason may be that the pretraining dataset is entirely focused on the SO - 100 robotic arm, and the overall VLA is relatively small.

FanZhang91, 2025-06-24:
> I think part of the reason may be that the pretraining dataset is entirely focused on the SO - 100 robotic arm, and the overall VLA is relatively small.

Thanks for your reply。 Is there any way to alleviate this problem?

zlw21gxy, 2025-07-04:
@FanZhang91 https://github.com/huggingface/lerobot/issues/1316#issuecomment-3026588850 this is exactly what I use

FanZhang91, 2025-07-04:
@zlw21gxy Thanks for your reply, but I think my overall pipeline seems functional, and I am not using the Libero dataset but my own data. The comment([1316](https://github.com/huggingface/lerobot/issues/1316#issuecomment-3026588850)) you mentioned above may not be of much help to me.

Rseelan, 2025-11-17:
Having the same issue as well...