huggingface/lerobot #1387: so101 can't work with HIL-SERL

yangfengzzz, 2025-06-26:
### System Info

```Shell
only so100_follower_end_effector can be used as robot. so when I use so-101, the ee limit is very small and can't used in human intervention. 

meanwhile, when device is cpu, 

"action": info["action_intervention"].cpu().squeeze(0).float() if policy is None else action

should be changed to:

"action": info["action_intervention"].astype('float32') if policy is None else action
```

### Information

- [x] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

follow docs to work rl with so-101

### Expected behavior

human intervention work well

YiChenCityU, 2025-08-01:
same problem

GaryyGitHub, 2025-08-07:
I also encountered the same problem. I think the issue should lie in the `find_joint_and_ee_bounds()` function of `find_joint_limits.py`. The action of so_101 represents the angles of each joint, while so100_follower_end_effector requires the changes in x, y, z. The wrong parameters were passed by `send_action()`, which led to the inability to perform teleoperation.

My solution is to add the following code in the `send_action()` method of `so100_follower_end_effector.py`: 

```
# Check if this is a joint space action (from teleoperator)
if isinstance(action, dict) and any(key.endswith('.pos') for key in action.keys()):
    # This is a joint space action, pass it directly to parent class
    return super().send_action(action)
```