huggingface/lerobot #1389: ConnectionError: Read failed due to communication error on port /dev/ttyACM0 for group_key Present_Position_shoulder_pan_shoulder_lift_elbow_flex_wrist_flex_wrist_roll_gripper: [TxRxResult] There is no status packet!

saraparoji, 2025-06-26:
Hello,
I have just encountered the problem

(lerobot) cvl@cvl-genesis:~/lerobot$ python3 lerobot/scripts/control_robot.py   --robot.type=so100   --robot.cameras='{}'   --control.type=teleoperate
INFO 2025-06-25 17:05:25 ol_robot.py:408 {'control': {'display_data': False, 'fps': None, 'teleop_time_s': None},
'robot': {'calibration_dir': '.cache/calibration/so100',
           'cameras': {},
           'follower_arms': {'main': {'mock': False,
                                      'motors': {'elbow_flex': [3, 'sts3215'],
                                                 'gripper': [6, 'sts3215'],
                                                 'shoulder_lift': [2,'sts3215'],
                                                 'shoulder_pan': [1, 'sts3215'],
                                                 'wrist_flex': [4, 'sts3215'],
                                                 'wrist_roll': [5, 'sts3215']},
                                      'port': '/dev/ttyACM0'}},
           'gripper_open_degree': None,
           'leader_arms': {'main': {'mock': False,
                                    'motors': {'elbow_flex': [3, 'sts3215'],
                                               'gripper': [6, 'sts3215'],
                                               'shoulder_lift': [2, 'sts3215'],
                                               'shoulder_pan': [1, 'sts3215'],
                                               'wrist_flex': [4, 'sts3215'],
                                               'wrist_roll': [5, 'sts3215']},
                                    'port': '/dev/ttyACM1'}},
           'max_relative_target': None,
           'mock': False}}
Connecting main follower arm.
Connecting main leader arm.
Activating torque on main follower arm.
INFO 2025-06-25 17:05:25 rol_utils.py:82 dt: 3.05 (328.1hz) dtRlead: 1.34 (745.5hz) dtWfoll: 0.16 (6393.3hz) dtRfoll: 1.50 (668.7hz)
INFO 2025-06-25 17:05:25 rol_utils.py:82 dt: 3.07 (326.2hz) dtRlead: 1.43 (699.1hz) dtWfoll: 0.13 (7756.0hz) dtRfoll: 1.49 (672.5hz)
INFO 2025-06-25 17:05:26 rol_utils.py:82 dt: 2.87 (348.1hz) dtRlead: 1.28 (782.1hz) dtWfoll: 0.09 (10675.1hz) dtRfoll: 1.49 (671.9hz)
Traceback (most recent call last):
  File "/home/cvl/lerobot/lerobot/scripts/control_robot.py", line 437, in <module>
    control_robot()
  File "/home/cvl/lerobot/lerobot/configs/parser.py", line 227, in wrapper_inner
    response = fn(cfg, *args, **kwargs)
  File "/home/cvl/lerobot/lerobot/scripts/control_robot.py", line 418, in control_robot
    teleoperate(robot, cfg.control)
  File "/home/cvl/lerobot/lerobot/common/robot_devices/utils.py", line 42, in wrapper
    raise e
  File "/home/cvl/lerobot/lerobot/common/robot_devices/utils.py", line 38, in wrapper
    return func(robot, *args, **kwargs)
  File "/home/cvl/lerobot/lerobot/scripts/control_robot.py", line 235, in teleoperate
    control_loop(
  File "/home/cvl/lerobot/lerobot/common/datasets/image_writer.py", line 36, in wrapper
    raise e
  File "/home/cvl/lerobot/lerobot/common/datasets/image_writer.py", line 29, in wrapper
    return func(*args, **kwargs)
  File "/home/cvl/lerobot/lerobot/common/robot_devices/control_utils.py", line 250, in control_loop
    observation, action = robot.teleop_step(record_data=True)
  File "/home/cvl/lerobot/lerobot/common/robot_devices/robots/manipulator.py", line 490, in teleop_step
    follower_pos[name] = self.follower_arms[name].read("Present_Position")
  File "/home/cvl/lerobot/lerobot/common/robot_devices/motors/feetech.py", line 751, in read
    raise ConnectionError(
ConnectionError: Read failed due to communication error on port /dev/ttyACM0 for group_key Present_Position_shoulder_pan_shoulder_lift_elbow_flex_wrist_flex_wrist_roll_gripper: [TxRxResult] There is no status packet!

I have already used Feetech's official debug tool and upgraded the firmware for all motors, but it did not resolve the issue. Can anyone help me with this issue?

qingx-cyber, 2025-09-10:
我也遇到同样的问题，你是怎么解决的？

ananth-pal, 2026-02-16:
@saraparoji was this from using a waveshare motor?

NikodemBartnik, 2026-10-29:
Thanks for the report and for sharing the firmware link. This was on the old `control_robot.py` stack, which has since been replaced: teleoperation is now `lerobot-teleoperate`, and the Feetech/Dynamixel code was rewritten into a single `SerialMotorsBus` (#4672).

If you still see "no status packet" errors, check the power supply voltage, the motor IDs (`lerobot-setup-motors`) and the cabling first, following the [SO-101 guide](https://huggingface.co/docs/lerobot/so101). I'm closing this as obsolete, but please open a fresh issue if it still happens with the latest release.

closed 2026-10-29 by NikodemBartnik