huggingface/lerobot #1410: GLFW Error

fireyan8, 2025-06-30:
Dear all,

when I tried to run:

`python lerobot/scripts/rl/gym_manipulator.py --config_path ./gym_js/config.json`

I will encounter:
```
F:\Program_Files\Anaconda\envs\lerobot\lib\site-packages\glfw\__init__.py:917: GLFWError: (65537) b'The GLFW library is not initialized'
  warnings.warn(message, GLFWError)
```

I am running windows11, with a nvidia GPU.

what should I do?

Thank you very mcuh.

raul-machine-learning, 2025-06-30:
try adding this at the top of your script

```
import os
os.environ["MUJOCO_GL"] = "egl"
os.environ["CUDA_VISIBLE_DEVICES"] = "0"
```

shs822, 2025-09-25:
> > try adding this at the top of your script
> > ```
> > import os
> > os.environ["MUJOCO_GL"] = "egl"
> > os.environ["CUDA_VISIBLE_DEVICES"] = "0"
> > ```
> 
> thank you I will try, i changed device to cpu. because the torch is not cuda supported.

can you run gymhil sim environment in Windows? I can't finish the code `pip install -e ".[hilserl]"`, it raise wrong(FileNotFoundError: [Errno 2] No such file or directory: 'project-config.jam') during installing build dependencies

pkooij, 2026-10-21:
Closing this older Windows setup report. Please reopen with current package versions, the command and full traceback if it still occurs.

closed 2026-10-21 by pkooij