huggingface/lerobot #1414: Identifying Motors 2-6

ramallis, 2025-06-30:
My configure_motor file only identifies the first motor, but not the rest of them. I am sending the following command: python lerobot/scripts/configure_motor.py   --port /dev/ttyACM0   --brand feetech   --model sts3215   --baudrate 1000000   --ID 2 (to the second motor of course), and after I do python -m lerobot.calibrate --robot.type so100_follower --robot.port /dev/ttyACM0 --robot.id my_follower_1 
and get RuntimeError: FeetechMotorsBus motor check failed on port '/dev/ttyACM0':

Missing motor IDs:
  - 2 (expected model: 777)
  - 3 (expected model: 777)
  - 4 (expected model: 777)
  - 5 (expected model: 777)
  - 6 (expected model: 777)

Full expected motor list (id: model_number):
{1: 777, 2: 777, 3: 777, 4: 777, 5: 777, 6: 777}

In this case, motor ID 2 should not be missing.

 Did anyone have that same issue? If so, how did you solve it?

chabir, 2025-07-02:
most likely, the cable between motor 1 and 2 isn't fully plugged.

Sankalpa-Raymondal, 2025-07-11:
Hi, I am also facing the same issue, but for motor ID 1 and 6 also. 
Background: Found the wire to the gripper for follower arm twisting around the motor. Unplugged the wire and fixed the twisting of wire issue, but facing Motor ID not found  issue now.

LetianLi, 2025-07-26:
oh im stupid, forgot to plug motor 1 back into the board after configuring all the ids.

shrutic12, 2025-07-27:
Hi @ramallis @LetianLi , did you sort the issue out? I have a similar issue with my follower robot. The error I get on trying to calibrate is :

RuntimeError:` FeetechMotorsBus motor check failed on port 'COM8':

Missing motor IDs:
  - 1 (expected model: 777)
  - 6 (expected model: 777)

Full expected motor list (id: model_number):
{1: 777, 2: 777, 3: 777, 4: 777, 5: 777, 6: 777}

Full found motor list (id: model_number):
{2: 777, 3: 777, 4: 777, 5: 777}
I have a pre-assembled so100 robot and taking the connections out for the motors is quite delicate to redo them. So I would like to avoid redoing the configure motor step. I checked the motor connections. All the motors light up as well so should be working. I am wondering what could cause the motor check fail and ideas to troubleshoot or work around this problem?

LetianLi, 2025-07-27:
The issue I encountered was just that i never plugged in motor 1 to the board, so the script just couldn't connect to motor 1 nor any of the other motors (so my full found motor list was empty).

If you said all 6 are lit up, then they should all be connected already. I would then suspect that maybe when setting up the motors you didn't make sure to disconnect the motor from other motors still in series because that can cause problems.

If you don't want to reset up all the motors, you could try and modify the script to only setup specific motors you want.
Script below untested but should work.
```python
# src/lerobot/robots/so100_follower/so100_follower.py
def setup_motors(self) -> None:
    print("Available motors:")
    for motor in reversed(self.bus.motors):
        print(f"  {self.bus.motors[motor].id}. {motor}")
    while True:
        motor_id = input("Enter the motor id to setup: ")
        motor = next((m for m in self.bus.motors if self.bus.motors[m].id == int(motor_id)), None)
        if motor:
            input(f"Connect the controller board to the '{motor}' motor only and press enter. Setting the id to {self.bus.motors[motor].id}")
            self.bus.setup_motor(motor)
            print(f"'{motor}' motor id set to {self.bus.motors[motor].id}")
            break
        else:
            print(f"Could not find motor with target id {motor_id}. Please try again.")
```

closed 2026-02-25 by github-actions[bot]