huggingface/lerobot #1438: lerobot training script is broken in google colab, referenced on the website

therealadityashankar, 2025-07-03:
### System Info

```Shell
n/a
```

### Information

- [x] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

I've raised it here, https://github.com/huggingface/notebooks/issues/603 but it is also connected to this project, so I am raising it here

This training script is also referenced on https://huggingface.co/docs/lerobot/notebooks, so new users will have an issue here

### Expected behavior

the training script should be correct

therealadityashankar, 2025-07-03:
So in the training colab at https://colab.research.google.com/github/huggingface/notebooks/blob/main/lerobot/training-act.ipynb

This is the corrected code, the training script broke, due to the lerobot github repository reorganizing itself

```
!cd lerobot && python lerobot/scripts/train.py \
  --dataset.repo_id=${HF_USER}/il_gym0 \
  --policy.type=act \
  --output_dir=outputs/train/il_sim_test0 \
  --job_name=il_sim_test \
  --policy.device=cuda \
  --wandb.enable=true
```

should instead be

```
!python lerobot/src/lerobot/scripts/train.py \
  --dataset.repo_id=${HF_USER}/jenga_dataset_2_video \
  --policy.type=act \
  --output_dir=outputs/train/il_sim_test0 \
  --job_name=il_sim_test \
  --policy.device=cuda \
  --policy.push_to_hub=false
```

doing-great-events, 2025-09-09:
Hi 
if I want to modify the steps, what should be the correct code,
I tried following code, but it fails, how can I modify the "steps" and "save_frequency"

!python train.py \
  --dataset.repo_id==${HF_USER}/so101-test \
  --policy.type=act \
  --log_freq=500 \
  --steps=8000  \
  --save_freq=1000  \
  --output_dir=outputs/train/run_resumption \
  --job_name=il_sim_test \
  --policy.device=cuda \
  --wandb.enable=False \
  --policy.repo_id=DGEs/my_policy \