huggingface/lerobot #1446: How to evaluate finetuned SmolVLA model

BintaoBryant, 2025-07-06:
Dear authors and your wonderful work.
I have fine-tuned the smolvla model based on a customized lerobot format dataset. My dataset is picking up a banana and placing it on a box. How can I evaluate the performance of the model? I tried eval.py in the scripes directory, but env_type=pusht doesn't work. I think this env_type may cause eval.py to fail to run.
I hope someone can help me. Thanks in advance.

zlw21gxy, 2025-07-07:
My understanding is that evaluation in a simulator isn't supported yet. You can monitor training and evaluation loss, but you cannot perform evaluations using the simulator at this time.

jadechoghari, 2025-10-17:
Hey ! To evaluate on a custom task you shoudld follow this guide: https://huggingface.co/docs/lerobot/en/il_robots#run-inference-and-evaluate-your-policy

Remember: we do support evaluation environemtns such as pusht, aloha, libero, but specify your env_type only if you actually trained on those environments. for real-world evals follow the above guide!