huggingface/lerobot #1456: Camera not detected on windows laptop

jeffrey-c-star, 2025-07-07:
Hello! I was wondering if anyone else ran into this issue or knows how to solve it. I am trying to run the record dataset script in windows anaconda prompt, but I keep getting this error with the computer not being able to detect the camera. I ran the script it suggested but it couldn't open the cameras (but was able to detect them). I made sure that the computer was able to open the camera through VSCode. 

If anyone can help, I would really appreciate it!

AoqunJin, 2025-07-07:
You can try to change this line:

https://github.com/huggingface/lerobot/blob/aec1b29d230341721c8d8f4413c47dcdb424cf5d/src/lerobot/cameras/opencv/camera_opencv.py#L158

To:

```python
self.videocapture = cv2.VideoCapture(self.index_or_path)
```

todateman, 2025-07-13:
I have submitted a pull request to fix this so that the backend can be switched correctly depending on the OS.

https://github.com/huggingface/lerobot/pull/1495