huggingface/lerobot #1492: Is there any plan to add a validation loss in the training pipeline, which is not dependent on simulation env.

mohitydv09, 2025-07-11:
Can we have a dataset split in the training code to run the model on a holdout validation episode to check loss on it?

PuzhenYuan, 2025-11-11:
same issue

milong26, 2025-12-30:
I want it too :)

By the way, is the process like this: collect 100 episodes (as an example), split them into 90 for training and 10 for validation. After training, do validation on these 10 episodes without connecting to a real robot or executing actions. Then, calculate the difference between the model’s output and the actual actions, which is validation loss?

milong26, 2025-12-30:
I wrote one calculate manually
<details>
<summary> train_and_eval.py</summary>

```python
"""
仿照lerobot_train写的带有validation功能的train
"""

#!/usr/bin/env python
import logging
import time
from contextlib import nullcontext
from pprint import pformat
from typing import Any

import torch
from accelerate import Accelerator
from termcolor import colored
from torch.optim import Optimizer

from lerobot.configs import parser
from lerobot.configs.train import TrainPipelineConfig
from lerobot.datasets.factory import make_dataset
from lerobot.datasets.sampler import EpisodeAwareSampler
from lerobot.datasets.utils import cycle
from lerobot.envs.factory import make_env, make_env_pre_post_processors
from lerobot.envs.utils import close_envs
from lerobot.optim.factory import make_optimizer_and_scheduler
from lerobot.policies.factory import make_policy, make_pre_post_processors
from lerobot.policies.pretrained import PreTrainedPolicy
from lerobot.rl.wandb_utils import WandBLogger
from lerobot.scripts.lerobot_eval import eval_policy_all
from lerobot.utils.import_utils import register_third_party_plugins
from lerobot.utils.logging_utils import AverageMeter, MetricsTracker
from lerobot.utils.random_utils import set_seed
from lerobot.utils.train_utils import (
    get_step_checkpoint_dir,
    get_step_identifier,
    load_training_state,
    save_checkpoint,
    update_last_checkpoint,
)
from lerobot.utils.utils import format_big_number, has_method, init_logging
from torch.utils.data import Subset
from typing import Tuple
import random

import torch.nn.functional as F  # noqa: N812
from lerobot.utils.constants import ACTION, OBS_ENV_STATE, OBS_IMAGES, OBS_STATE

# 划分数据集，按 episode 划分
def split_dataset(dataset, train_ratio: float = 0.8, seed: int = 42) -> Tuple[Subset, Subset]:
    """
    Split an episodic dataset into train and validation sets by episode.

    Args:
        dataset: Dataset object, expected to have `meta.episodes` dict with keys 'dataset_from_index' and 'dataset_to_index'
        train_ratio: Fraction of episodes to use for training (0 < train_ratio <= 1)
        seed: Random seed for reproducibility

    Returns:
        train_dataset, val_dataset: two Subset instances
    """
    if not hasattr(dataset, "meta") or not hasattr(dataset.meta, "episodes"):
        raise ValueError("Dataset must have meta.episodes for episodic splitting")

    episodes_meta = dataset.meta.episodes  # dict with keys: dataset_from_index, dataset_to_index
    num_episodes = len(episodes_meta["dataset_from_index"])
    indices = list(range(num_episodes))
[... 187 lines cut ...]
                        F.l1_loss(val_batch[ACTION], actions_hat, reduction="none")
                        * ~val_batch["action_is_pad"].unsqueeze(-1)
                    ).mean()

                    val_tracker.loss = l1_loss.item()
                    val_tracker.step()

            if wandb_logger:
                wandb_logger.log_dict(val_tracker.to_dict(), step, mode="eval")

            policy.model.train()
            policy.train()

def main():
    register_third_party_plugins()
    train_and_validate()

if __name__ == "__main__":
    main()

```

</details>

In code eval_frequency is defined in config/train.py