huggingface/lerobot #1499: Loading video indefinitely

drain-pipe, 2025-07-13:
After starting a training script the terminal just outputs this for about half an hour before I stop it, is this normal? Seems like an issue..

```
Loading video for ep_idx=21, vid_key=observation.images.front
Video path: /home/ubuntu/.cache/huggingface/lerobot/DrainpipeAI/stockv1/videos/chunk-000/observation.images.front/episode_000021.mp4
Loading video for ep_idx=21, vid_key=observation.images.side
Video path: /home/ubuntu/.cache/huggingface/lerobot/DrainpipeAI/stockv1/videos/chunk-000/observation.images.side/episode_000021.mp4
Loading video for ep_idx=11, vid_key=observation.images.front
Video path: /home/ubuntu/.cache/huggingface/lerobot/DrainpipeAI/stockv1/videos/chunk-000/observation.images.front/episode_000011.mp4
Loading video for ep_idx=11, vid_key=observation.images.side
Video path: /home/ubuntu/.cache/huggingface/lerobot/DrainpipeAI/stockv1/videos/chunk-000/observation.images.side/episode_000011.mp4
Loading video for ep_idx=21, vid_key=observation.images.front
Video path: /home/ubuntu/.cache/huggingface/lerobot/DrainpipeAI/stockv1/videos/chunk-000/observation.images.front/episode_000021.mp4
Loading video for ep_idx=21, vid_key=observation.images.side
Video path: /home/ubuntu/.cache/huggingface/lerobot/DrainpipeAI/stockv1/videos/chunk-000/observation.images.side/episode_000021.mp4
Loading video for ep_idx=50, vid_key=observation.images.front
Video path: /home/ubuntu/.cache/huggingface/lerobot/DrainpipeAI/stockv1/videos/chunk-000/observation.images.front/episode_000050.mp4
Loading video for ep_idx=50, vid_key=observation.images.side
Video path: /home/ubuntu/.cache/huggingface/lerobot/DrainpipeAI/stockv1/videos/chunk-000/observation.images.side/episode_000050.mp4
Loading video for ep_idx=34, vid_key=observation.images.front
Video path: /home/ubuntu/.cache/huggingface/lerobot/DrainpipeAI/stockv1/videos/chunk-000/observation.images.front/episode_000034.mp4
Loading video for ep_idx=34, vid_key=observation.images.side
Video path: /home/ubuntu/.cache/huggingface/lerobot/DrainpipeAI/stockv1/videos/chunk-000/observation.images.side/episode_000034.mp4
```

Saurav1700, 2025-07-22:
Did you solve it?

drain-pipe, 2025-07-22:
> Did you solve it?

I had my Conda environment set up incorrectly. I used uv instead as suggested by https://docs.phospho.ai/learn/train-smolvla, which worked fine.