huggingface/lerobot #1500: async inference doesn't generate actions.Observation #0 has been filtered out

milong26, 2025-07-13:
I tried to use async inference,

I run this in server: `python -m lerobot.scripts.server.policy_server     --host="xx,xx,xx,xx"     --port=9000`
And have models in server.

run this locally:
`PYTHONPATH=src python src/lerobot/scripts/server/robot_client.py --config_path=async_evaluate/template.yaml`

the yaml looks like:
```yaml

server_address: xx.xx.xx.xx:9000

robot:
  type: so100_follower
  port: /dev/ttyACM0
  id: congbi
  cameras:
    side:
      type: intelrealsense
      serial_number_or_name: aaaaaaaaaaa
      width: 640
      height: 480
      fps: 30
      use_depth: false
    wrist:
      type: opencv
      index_or_path: 6
      width: 640
      height: 480
      fps: 30

task: "pick up the cube and place it into the box."
policy_type: smolvla
pretrained_name_or_path: outputs/train/bbb/checkpoints/last/pretrained_model
policy_device: cuda

actions_per_chunk: 50
chunk_size_threshold: 0.5
aggregate_fn_name: weighted_average
debug_visualize_queue_size: true

# dataset:
#   repo_id: eval/eval_firts50_baseline
#   num_episodes: 1
#   episode_time_s: 60
#   push_to_hub: false
#   reset_time_s: 0

```

Then I saw these in server(they look similar)
```
INFO 2025-07-13 12:59:54 y_server.py:175 Received observation #0 | Avg FPS: 5.95 | Target: 30.00 | One-way latency: 2.17ms
INFO 2025-07-13 12:59:54 y_server.py:191 Observation #0 has been filtered out
INFO 2025-07-13 12:59:54 /helpers.py:354  Starting receiver
INFO 2025-07-13 12:59:54 y_server.py:175 Received observation #0 | Avg FPS: 5.95 | Target: 30.00 | One-way latency: 1.89ms
INFO 2025-07-13 12:59:54 y_server.py:191 Observation #0 has been filtered out
INFO 2025-07-13 12:59:54 /helpers.py:354  Starting receiver
INFO 2025-07-13 12:59:54 y_server.py:175 Received observation #0 | Avg FPS: 5.95 | Target: 30.00 | One-way latency: 1.80ms
INFO 2025-07-13 12:59:54 y_server.py:191 Observation #0 has been filtered out
INFO 2025-07-13 12:59:54 /helpers.py:354  Starting receiver
INFO 2025-07-13 12:59:54 y_server.py:175 Received observation #0 | Avg FPS: 5.95 | Target: 30.00 | One-way latency: 2.52ms
INFO 2025-07-13 12:59:54 y_server.py:191 Observation #0 has been filtered out
INFO 2025-07-13 12:59:54 /helpers.py:354  Starting receiver
INFO 2025-07-13 12:59:55 y_server.py:175 Received observation #0 | Avg FPS: 5.95 | Target: 30.00 | One-way latency: 2.25ms
INFO 2025-07-13 12:59:55 y_server.py:191 Observation #0 has been filtered out
INFO 2025-07-13 12:59:55 /helpers.py:354  Starting receiver
INFO 2025-07-13 12:59:55 y_server.py:175 Received observation #0 | Avg FPS: 5.95 | Target: 30.00 | One-way latency: 2.10ms
INFO 2025-07-13 12:59:55 y_server.py:191 Observation #0 has been filtered out
INFO 2025-07-13 12:59:55 /helpers.py:354  Starting receiver
INFO 2025-07-13 12:59:55 y_server.py:175 Received observation #0 | Avg FPS: 5.95 | Target: 30.00 | One-way latency: 1.77ms
INFO 2025-07-13 12:59:55 y_server.py:191 Observation #0 has been filtered out
INFO 2025-07-13 12:59:55 /helpers.py:354  Starting receiver
INFO 2025-07-13 12:59:55 y_server.py:175 Received observation #0 | Avg FPS: 5.95 | Target: 30.00 | One-way latency: 1.81ms
INFO 2025-07-13 12:59:55 y_server.py:191 Observation #0 has been filtered out
INFO 2025-07-13 12:59:55 /helpers.py:354  Starting receiver
INFO 2025-07-13 12:59:55 y_server.py:175 Received observation #0 | Avg FPS: 5.95 | Target: 30.00 | One-way latency: 1.92ms
INFO 2025-07-13 12:59:55 y_server.py:191 Observation #0 has been filtered out
INFO 2025-07-13 12:59:55 /helpers.py:354  Starting receiver
INFO 2025-07-13 12:59:55 y_server.py:175 Received observation #0 | Avg FPS: 5.95 | Target: 30.00 | One-way latency: 1.84ms
INFO 2025-07-13 12:59:55 y_server.py:191 Observation #0 has been filtered out
INFO 2025-07-13 12:59:55 /helpers.py:354  Starting receiver
```
By the way I saw `reduce layer to 16` so the server got policy message and load policy correcly.

And to test I add an Error om `getAction` function 
```
            # Create and return the action chunk
            actions = async_inference_pb2.Actions(data=actions_bytes)
            raise KeyError("generate an action")
            self.logger.info(
                f"Action chunk #{obs.get_timestep()} generated | "
                f"Total time: {(inference_time + serialize_time) * 1000:.2f}ms"
            )
```
But nothing happened. So I think getAction wasn't used at all.

And locally terminal said:
```
INFO 2025-07-13 20:59:54 t_client.py:471 Control loop (ms): 166.87
INFO 2025-07-13 20:59:55 t_client.py:213 Sent observation #0 | 
INFO 2025-07-13 20:59:55 t_client.py:471 Control loop (ms): 168.18
INFO 2025-07-13 20:59:55 t_client.py:213 Sent observation #0 | 
INFO 2025-07-13 20:59:55 t_client.py:471 Control loop (ms): 169.37
INFO 2025-07-13 20:59:55 t_client.py:213 Sent observation #0 | 
INFO 2025-07-13 20:59:55 t_client.py:471 Control loop (ms): 165.41
INFO 2025-07-13 20:59:55 t_client.py:213 Sent observation #0 | 
INFO 2025-07-13 20:59:55 t_client.py:471 Control loop (ms): 165.57
INFO 2025-07-13 20:59:55 t_client.py:213 Sent observation #0 | 
INFO 2025-07-13 20:59:55 t_client.py:471 Control loop (ms): 166.10
```
(My server time is incorrect, I run them at the same time.)

And of course no action made by my robot.;-;

How to solve it?

lerobot: d2645cb19fc521e5b117fe03d90a84f698d3d3f6

closed 2025-07-30 by fracapuano

kk2kl, 2025-09-09:
> I tried to use async inference,
> 
> I run this in server: `python -m lerobot.scripts.server.policy_server --host="xx,xx,xx,xx" --port=9000` And have models in server.
> 
> run this locally: `PYTHONPATH=src python src/lerobot/scripts/server/robot_client.py --config_path=async_evaluate/template.yaml`
> 
> the yaml looks like:
> 
> server_address: xx.xx.xx.xx:9000
> 
> robot:
>   type: so100_follower
>   port: /dev/ttyACM0
>   id: congbi
>   cameras:
>     side:
>       type: intelrealsense
>       serial_number_or_name: aaaaaaaaaaa
>       width: 640
>       height: 480
>       fps: 30
>       use_depth: false
>     wrist:
>       type: opencv
>       index_or_path: 6
>       width: 640
>       height: 480
>       fps: 30
> 
> task: "pick up the cube and place it into the box."
> policy_type: smolvla
> pretrained_name_or_path: outputs/train/bbb/checkpoints/last/pretrained_model
> policy_device: cuda
> 
> actions_per_chunk: 50
> chunk_size_threshold: 0.5
> aggregate_fn_name: weighted_average
> debug_visualize_queue_size: true
> 
> 
> # dataset:
> #   repo_id: eval/eval_firts50_baseline
> #   num_episodes: 1
> #   episode_time_s: 60
> #   push_to_hub: false
> #   reset_time_s: 0
> Then I saw these in server(they look similar)
> 
> ```
> INFO 2025-07-13 12:59:54 y_server.py:175 Received observation #0 | Avg FPS: 5.95 | Target: 30.00 | One-way latency: 2.17ms
> INFO 2025-07-13 12:59:54 y_server.py:191 Observation #0 has been filtered out
> INFO 2025-07-13 12:59:54 /helpers.py:354  Starting receiver
> INFO 2025-07-13 12:59:54 y_server.py:175 Received observation #0 | Avg FPS: 5.95 | Target: 30.00 | One-way latency: 1.89ms
> INFO 2025-07-13 12:59:54 y_server.py:191 Observation #0 has been filtered out
> INFO 2025-07-13 12:59:54 /helpers.py:354  Starting receiver
> INFO 2025-07-13 12:59:54 y_server.py:175 Received observation #0 | Avg FPS: 5.95 | Target: 30.00 | One-way latency: 1.80ms
> INFO 2025-07-13 12:59:54 y_server.py:191 Observation #0 has been filtered out
> INFO 2025-07-13 12:59:54 /helpers.py:354  Starting receiver
> INFO 2025-07-13 12:59:54 y_server.py:175 Received observation #0 | Avg FPS: 5.95 | Target: 30.00 | One-way latency: 2.52ms
> INFO 2025-07-13 12:59:54 y_server.py:191 Observation #0 has been filtered out
> INFO 2025-07-13 12:59:54 /helpers.py:354  Starting receiver
> INFO 2025-07-13 12:59:55 y_server.py:175 Received observation #0 | Avg FPS: 5.95 | Target: 30.00 | One-way latency: 2.25ms
> INFO 2025-07-13 12:59:55 y_server.py:191 Observation #0 has been filtered out
> INFO 2025-07-13 12:59:55 /helpers.py:354  Starting receiver
> INFO 2025-07-13 12:59:55 y_server.py:175 Received observation #0 | Avg FPS: 5.95 | Target: 30.00 | One-way latency: 2.10ms
> INFO 2025-07-13 12:59:55 y_server.py:191 Observation #0 has been filtered out
> INFO 2025-07-13 12:59:55 /helpers.py:354  Starting receiver
> INFO 2025-07-13 12:59:55 y_server.py:175 Received observation #0 | Avg FPS: 5.95 | Target: 30.00 | One-way latency: 1.77ms
> INFO 2025-07-13 12:59:55 y_server.py:191 Observation #0 has been filtered out
> INFO 2025-07-13 12:59:55 /helpers.py:354  Starting receiver
> INFO 2025-07-13 12:59:55 y_server.py:175 Received observation #0 | Avg FPS: 5.95 | Target: 30.00 | One-way latency: 1.81ms
> INFO 2025-07-13 12:59:55 y_server.py:191 Observation #0 has been filtered out
> INFO 2025-07-13 12:59:55 /helpers.py:354  Starting receiver
> INFO 2025-07-13 12:59:55 y_server.py:175 Received observation #0 | Avg FPS: 5.95 | Target: 30.00 | One-way latency: 1.92ms
> INFO 2025-07-13 12:59:55 y_server.py:191 Observation #0 has been filtered out
> INFO 2025-07-13 12:59:55 /helpers.py:354  Starting receiver
> INFO 2025-07-13 12:59:55 y_server.py:175 Received observation #0 | Avg FPS: 5.95 | Target: 30.00 | One-way latency: 1.84ms
> INFO 2025-07-13 12:59:55 y_server.py:191 Observation #0 has been filtered out
> INFO 2025-07-13 12:59:55 /helpers.py:354  Starting receiver
> ```
> 
> By the way I saw `reduce layer to 16` so the server got policy message and load policy correcly.
> 
> And to test I add an Error om `getAction` function
> 
> ```
>             # Create and return the action chunk
>             actions = async_inference_pb2.Actions(data=actions_bytes)
>             raise KeyError("generate an action")
>             self.logger.info(
>                 f"Action chunk #{obs.get_timestep()} generated | "
>                 f"Total time: {(inference_time + serialize_time) * 1000:.2f}ms"
>             )
> ```
> 
> But nothing happened. So I think getAction wasn't used at all.
> 
> And locally terminal said:
> 
> ```
> INFO 2025-07-13 20:59:54 t_client.py:471 Control loop (ms): 166.87
> INFO 2025-07-13 20:59:55 t_client.py:213 Sent observation #0 | 
> INFO 2025-07-13 20:59:55 t_client.py:471 Control loop (ms): 168.18
> INFO 2025-07-13 20:59:55 t_client.py:213 Sent observation #0 | 
> INFO 2025-07-13 20:59:55 t_client.py:471 Control loop (ms): 169.37
> INFO 2025-07-13 20:59:55 t_client.py:213 Sent observation #0 | 
> INFO 2025-07-13 20:59:55 t_client.py:471 Control loop (ms): 165.41
> INFO 2025-07-13 20:59:55 t_client.py:213 Sent observation #0 | 
> INFO 2025-07-13 20:59:55 t_client.py:471 Control loop (ms): 165.57
> INFO 2025-07-13 20:59:55 t_client.py:213 Sent observation #0 | 
> INFO 2025-07-13 20:59:55 t_client.py:471 Control loop (ms): 166.10
> ```
> 
> (My server time is incorrect, I run them at the same time.)
> 
> And of course no action made by my robot.;-;
> 
> How to solve it?
> 
> lerobot: [d2645cb](https://github.com/huggingface/lerobot/commit/d2645cb19fc521e5b117fe03d90a84f698d3d3f6)

@milong26 may I ask how you managed to solve this error please? I am facing the same error now, the robot does not generate action when running policy_server and robot_client.py

kk2kl, 2025-09-09:
> Thanks! It runs correctly now. If anyone met same problem here is my experience:
> 
> 1. check the first observation `must_go` ,it should be true
> 2. if it is, `_enqueue_observation` should return true.
> 3. function `GetActions` should predict action chunks and send to local
> 4. in `robotclient.py`,`send_action` should send action instruction to robot.
> 
> But I find local send action is less smooth than not using async inference. I will test later.
> 
> By the way. Do I need to stop server when I stopped locally? Since when I first run
> 
> ```
> server listening ...
> local def start(self): send model config to server -> server local model ->
> local send observation to server -> server get obs -> server predict action chunk -> server send action chunk to local -> local send action to robot
> ...
> local stopped.
> ```
> 
> if I don't stop server, will the server reload model again?

@milong26 could you explain in more details how you implement these in your code please?

milong26, 2025-09-09:
@kk2kl Try to run the server for longer? From your logs (thank you so much for adding them!) it looks like you're running the server (minus setup) for ~1s only this method works.

Or you can put your object to another place, since in one trial, I saw action generated but arm didn't work.
There're two other methods which may help u check

1. write a function(such as listen to keyboard) to clear action queues and set `must_go` to true.
2. watch the output of policy_server and robot client. Make sure
    1. first observation must_go =true, server should generate some actions and sent them to client.
    2. client pick actions from queue one bye one, and robot follow the action.
    3. there should be some like policy_server_1757343164.log in log/ folder.

By the way my function looks like(you can ask ai to write one too)
```python

# in init
        # 监听键盘输入->的时候清空action缓存
        threading.Thread(target=self._listen_clear_key,daemon=True).start()
        # 等待时间
        self.pause_until=0
        self.use_skip_step=config.use_skip_step

# listen to keyboard,in robot_clinet.py
    def _listen_clear_key(self):
        def on_press(key):
            try:
                if key==keyboard.Key.right:
                    self.clear_action_queue()
                if key==keyboard.Key.left:
                    self.running=False
            except AttributeError:
                self.logger.info("键盘出问题")
        with keyboard.Listener(on_press=on_press) as listener:
            listener.join()

    def clear_action_queue(self,pause_seconds:float=3.0):
        with self.action_queue_lock:
            # while not self.action_queue.empty():
                # self.action_queue.get_nowait()
                # 直接新建队列
                self.action_queue = Queue()
        self.must_go.set()
        self.logger.info("清空了actionqueue，顺便设置了mustgo，停个几秒")
        # 要设置control_loop里面暂停
        self.pause_until=time.time()+pause_seconds

    def control_loop(self, task: str, verbose: bool = False) -> tuple[Observation, Action]:
        """Combined function for executing actions and streaming observations"""
        # Wait at barrier for synchronized start
        self.start_barrier.wait()
        self.logger.info("Control loop thread starting")

        _performed_action = None
        _captured_observation = None

        while self.running:

            # 右键清空，暂停几秒
            if time.time()<self.pause_until:
                time.sleep(0.1)
                continue

```

milong26, 2025-09-18:
@kk2kl lerobot hasn't written distributed training till now. see https://github.com/huggingface/lerobot/issues/1666 luckily it said it will go on after dataset v3 and i saw dataset v3 is updated. 

I ruined the `torch.cuda.is_available()` when I didn't set CUDA_VISIBLE_DEVICES=0/1/2 before so I only tried using single gpu to train and evaluate. You can search how others finish this.

yylawlaw, 2025-09-19:
@milong26     thanks for sharing the info, looking forward to see more interesting features created by lerobot team and community.