huggingface/lerobot #1507: [PI0] Evaluation result on the metaworld

chenkang455, 2025-07-14:
Has anyone tried training pi0 on the Metaworld benchmark? My evaluation results are relatively low 30~%.

chenkang455, 2025-07-19:
> how you train your model, full-tune?

No, I use the train_expert_only setting. Results show that my model might over-fit...

https://github.com/user-attachments/assets/fed213de-7ce2-43d0-8562-6d5fcd82a84d

chendust, 2025-07-31:
@chenkang455 hi，bro。Can you provide the evaluation code for the metaworld dataset? Or could you tell me if there are any reference materials available? It seems that different evaluation methods can lead to significantly different final results.

pkooij, 2025-10-08:
Hi we just released an updated Pi0 and newly introduced pi05 implementation, we will also release meta world integration in LeRobot soon. Closing this issue for now, please reopen if you have any other questions.

closed 2025-10-08 by pkooij