huggingface/lerobot #1516: pi0fast

Hale-Sage, 2025-07-16:
When running pi0fast, I encountered the following error. How can I resolve it?

gribok201, 2025-07-17:
Try this:
pip uninstall -y transformers && pip cache purge
pip install "transformers==4.46.3" "accelerate>=1.3"

python - <<'PY'
from transformers import PaliGemmaConfig, __version__ as v
print("Transformers", v)
print("ignore_index =", PaliGemmaConfig().ignore_index)   # should print -100
PY

If you see ignore_index = -100, rerun your original python scripts/train.py … command;
the AttributeError should be gone.

Hale-Sage, 2025-07-22:
> Try this: pip uninstall -y transformers && pip cache purge pip install "transformers==4.46.3" "accelerate>=1.3"
>
> python - <<'PY' from transformers import PaliGemmaConfig, **version** as v [...]

Thanks, this is useful to me.

closed 2025-07-22 by Hale-Sage

miya-mizumitsuha, 2025-09-10:
When evaluating pi0fast model pre-trained on lerobot/metaworld_mt50, I meet the same problem 
AttributeError: 'GemmaModel' object has no attribute 'prepare_inputs_for_generation'; 

I follow the suggestion above to downgrade transformers to 4.46.3, but encounter many other problems during training due to the version mismatch. I believe current lerobot-main branch requires 4.52+.

Is there any other possible solution to this problem?